espressif/ESP8266_RTOS_SDK #17: Initializing multicast socket crash

qdk0901, 2015-01-30:
Hi, 
  I use below code to test receiving multicast message, but the add membership operation will crash.
This test code working fine on PC.

btw, make sure 
#define LWIP_IGMP                       1
in lwip/opt.h, otherwise compilation will failed

unsigned char ssdp_buf[1024];
void ssdp()
{
	struct sockaddr_in si = {0};
	int ssdp_fdm;
	int one = 1;
	struct ip_mreq imr;
	
	ssdp_fdm = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);
	if (ssdp_fdm < 0) {
		printf("create server socket failed\n");
		return;
	}
	
	setsockopt(ssdp_fdm, SOL_SOCKET, SO_REUSEADDR, &one, sizeof(int));
	
	si.sin_family = AF_INET;
	si.sin_port = htons(1900);

	if(bind(ssdp_fdm, (struct sockaddr *)&si, sizeof(struct sockaddr_in)) < 0) {
		close(ssdp_fdm);
		return;
	}
	
	memset(&imr, 0, sizeof(imr));
	imr.imr_multiaddr.s_addr = inet_addr("239.255.255.250");
	
       //add membership operation crash the ESP8266
	if(setsockopt(ssdp_fdm, IPPROTO_IP, IP_ADD_MEMBERSHIP, &imr, 
		sizeof(struct ip_mreq)) < 0) {
		close(ssdp_fdm);
		return;
	}
	
	printf("start listen\n");
	
	while(1) {
		int n = read(ssdp_fdm, ssdp_buf, 1024);
		if (n <= 0)
			break;
		printf("%s\n", ssdp_buf);
	}
	
	close(ssdp_fdm);
}

wizhippo, 2015-01-30:
Are you sure it is this call? When I have attempted to recompile lwip whenever the sta dhcp started I would get a crash as have others

https://github.com/espressif/esp_iot_rtos_sdk/issues/14

qdk0901, 2015-01-30:
Finally I have figured out that the default liblwip.a built by espressif has no IGMP support, then I tried build a lwip with IGMP myself, but got the same result as you.

closed 2018-09-18 by FayeY