espressif/ESP8266_RTOS_SDK #21: system_get_sdk_version not found

tve, 2015-04-04:
Using the new v0.9.9 I get the following trying to link the unchanged sample app:
```
user/.output/eagle/debug/lib/libuser.a(user_main.o):(.irom0.text+0x2c): undefined reference to `system_get_sdk_version'
```
The string isn't found in any library:
```
$ grep -r system_get_sdk_version .
./app/user/user_main.c:    //printf("SDK version:%s\n", system_get_sdk_version());
./bin/eagle.S:    //printf("SDK version:%s\n", system_get_sdk_version());
./bin/eagle.S:    //printf("SDK version:%s\n", system_get_sdk_version());
./include/espressif/esp_system.h:const char* system_get_sdk_version(void);
```

Did you perhaps forget to upload the new libraries to esp_iot_rtos_sdk_lib?

MacDada, 2015-04-17:
I probably should have reported it here, not there: https://github.com/mziwisky/esp8266-dev/issues/5

MacDada, 2015-04-17:
If this is a correct `make` result, than it works ;-)

```
git remote add tve https://github.com/tve/esp_iot_rtos_sdk.git
git fetch tve
git checkout tve/master
```

```
vagrant@precise32:/opt/Espressif/esp8266_rtos_sdk$ make clean
make -C app clean;
make[1]: Entering directory `/opt/Espressif/esp8266_rtos_sdk/app'
make -C user clean;
make[2]: Entering directory `/opt/Espressif/esp8266_rtos_sdk/app/user'
rm -f -r .output/eagle/debug
make[2]: Leaving directory `/opt/Espressif/esp8266_rtos_sdk/app/user'
rm -f -r .output/eagle/debug
make[1]: Leaving directory `/opt/Espressif/esp8266_rtos_sdk/app'
rm -f -r .output//
vagrant@precise32:/opt/Espressif/esp8266_rtos_sdk$ make
make[1]: Entering directory `/opt/Espressif/esp8266_rtos_sdk/app'
make[2]: Entering directory `/opt/Espressif/esp8266_rtos_sdk/app/user'
DEPEND: xtensa-lx106-elf-gcc -M -Os -g -O2 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -DICACHE_FLASH -I include -I ../include -I ../../include -I ../../extra_include -I ../../include/lwip -I ../../include/lwip/ipv4 -I ../../include/lwip/ipv6 -I ../../include/espressif user_main.c
make[2]: Leaving directory `/opt/Espressif/esp8266_rtos_sdk/app/user'
make[2]: Entering directory `/opt/Espressif/esp8266_rtos_sdk/app/user'
xtensa-lx106-elf-gcc -Os -g -O2 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals  -DICACHE_FLASH   -I include -I ../include -I ../../include -I ../../extra_include -I ../../include/lwip -I ../../include/lwip/ipv4 -I ../../include/lwip/ipv6 -I ../../include/espressif  -o .output/eagle/debug/obj/user_main.o -c user_main.c
xtensa-lx106-elf-ar ru .output/eagle/debug/lib/libuser.a .output/eagle/debug/obj/user_main.o 
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/libuser.a
make[2]: Leaving directory `/opt/Espressif/esp8266_rtos_sdk/app/user'
xtensa-lx106-elf-gcc  -L../lib -nostdlib -T../ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lmain -lfreertos -llwip -ludhcp user/.output/eagle/debug/lib/libuser.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 

!!!
COMPILE=gcc python ../tools/gen_appbin.py .output/eagle/debug/image/eagle.app.v6.out 0 0 0 0
No boot needed.
Generate eagle.flash.bin and eagle.irom0text.bin successully in folder bin.
eagle.flash.bin-------->0x00000
eagle.irom0text.bin---->0x40000
!!!
make[1]: Leaving directory `/opt/Espressif/esp8266_rtos_sdk/app'
```

closed 2018-09-18 by FayeY