espressif/ESP8266_RTOS_SDK #29: hw_timer based on NMI does not exist in RTOS SDK?

rhaenel, 2015-07-20:
I'm looking for a way to implement the hw_timer within the NMI. This is available on the non-OS SDK (e.g. void hw_timer_init use with NMI_SOURCE).

Unfortunately it seems that this does not work with the RTOS SDK, and I cannot fix it by just copying some header files because some functions like NmiTimSetFunc are missing from libmain.a.

The reasoning behind NMI timers is that I have a very time-critical code which must not be interrupted by any ISR or even the NMI. By executing this code from within the NMI, I would have a way to guarantee that really nothing can interrupt me there.

Any help is greatly appreciated.

magres, 2017-04-15:
Hi, I have the same question, have you got any solution?

donghengqaz, 2018-09-18:
This Function may be supported at later version, to be waiting please.

FayeY, 2018-10-23:
Hi, so sorry that NMI is used by the Wi-Fi low layer, cannot be used by hw_timer. Sorry for the inconvenience.

closed 2018-10-23 by FayeY