espressif/ESP8266_RTOS_SDK #31: multicast setsockopt fail, errno = 92( Protocol not available)

cccxxxttt, 2015-08-13:
Hi,
this is a old question. I user bellow code to create a udp multicast server, but join a mcast group failed.
the test code is find to PC.

       #define LWIP_IGMP 1         // first, make LWIP_IGMP is 1

	// join a mcast group
	struct ip_mreq mreq;
	mreq.imr_multiaddr.s_addr = inet_addr(MCAST_ADDR);	// mcast group ip
	mreq.imr_interface.s_addr = htonl(INADDR_ANY);      // local ip
	ret = setsockopt(sock_fd,IPPROTO_IP,IP_ADD_MEMBERSHIP,&mreq,sizeof(mreq));
	if(ret < 0) {
		WARNING("setsockopt() fail-%d:IP ADD MEMBURSHIP", errno);
//		goto __exit;
	}

the debug msg printf  "errno = 92", Macro definition as follows:
#define  ENOPROTOOPT  92  /* Protocol not available */

Hope that the engineer can solve this problem, Thank you very much.

wizhippo, 2015-08-13:
Did you rebuild lwip?  As you pointed out LWIP_IGMP is not set in the prebuilt bins.

cccxxxttt, 2015-08-20:
yes, i rebuid it.  Now I'am use udp broadcast to find device,  feel that  efficiency is not high.

closed 2018-09-18 by FayeY