espressif/ESP8266_RTOS_SDK #38: build fail: build example smart_config fail

quant67, 2015-11-13:
I just run gen_musc.sh in example/smart_config/ as default but  it fail after print massage like this:
```
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
```
I had build project_template and it become success.
My platform is a given virtual mirror ESP8266_lubuntu_20141021.ova. Any advice to me to solve this?

baoshi, 2015-11-13:
Open ld/eagle.app.v6.ld, read the notice and modify irom0 len according to your module's flash size

quant67, 2015-11-13:
Done! thanks.

closed 2015-11-13 by quant67