espressif/ESP8266_RTOS_SDK #46: always return -1 when invoke function: connect

suda-morris, 2015-12-29:
Here is my code, based on RTOS SDK1.3, the program works well except TCP client, what's wrong with it?
```C 
#include "esp_common.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "lwip/sockets.h"

#define SERVER_IP		"192.168.1.121"
#define SERVER_PORT		5000
#define TCP_DATA_LEN	256

static int32 sock_fd;

void tcpc_task() {
	int32 ret = 0;
	int32 recbytes;
	char *recv_buf;
	struct sockaddr_in tcpServerAddr;

	memset(&tcpServerAddr, 0, sizeof(tcpServerAddr));
	tcpServerAddr.sin_family = AF_INET;
	tcpServerAddr.sin_addr.s_addr = inet_addr(SERVER_IP);
	tcpServerAddr.sin_len = sizeof(tcpServerAddr);
	tcpServerAddr.sin_port = htons(SERVER_PORT);
	while (1) {
		sock_fd = socket(AF_INET, SOCK_STREAM, 0);
		if (sock_fd == -1) {
			printf("ESP8266 TCP client task > socket error\r\n");
			close(sock_fd);
			vTaskDelay(1000 / portTICK_RATE_MS);
			continue;
		}
		printf("ESP8266 TCP client task > create socket: %d\r\n", sock_fd);

		ret = connect(sock_fd, (struct sockaddr* )&tcpServerAddr,
				sizeof(struct sockaddr));
		if (ret != 0) {
			printf("ESP8266 TCP client task > connect fail!\r\n");
			close(sock_fd);
			vTaskDelay(1000 / portTICK_RATE_MS);
			continue;
		}
		printf("ESP8266 TCP client task > connect ok!\r\n");

		char *recv_buf = (char *) zalloc(TCP_DATA_LEN);
		while ((recbytes = read(sock_fd, recv_buf, TCP_DATA_LEN)) > 0) {
			recv_buf[recbytes] = 0;
			printf("ESP8266 TCP client task > recv data %d bytes!\r\n",
					recbytes);
			write(sock_fd, recv_buf, strlen(recv_buf) + 1);
		}
		free(recv_buf);
		if (recbytes <= 0) {
			printf("ESP8266 TCP client task > read data fail!\n");
			close(sock_fd);
		}
	}
}
```

Eszartek, 2015-12-29:
Here is a complete working example based on your code. I set it to connect to a public ftp server for demo purposes. Note that I changed the location of your zalloc since I was getting an exception using your placement (probably due to proximity of the network i/o code). I threw in a few delays here & there so it won't pound the server to hard.

```
#include <esp_common.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <uart.h>
#include "lwip/sockets.h"

#define SERVER_IP       "90.130.70.73"
#define SERVER_PORT     21
#define TCP_DATA_LEN    256

static int32 sock_fd;
void tcpc_task() {
	int32 ret = 0;
	int32 recbytes;
	char *recv_buf = (char *)malloc(TCP_DATA_LEN);
	struct sockaddr_in tcpServerAddr;
	
	memset(&tcpServerAddr, 0, sizeof(tcpServerAddr));
	tcpServerAddr.sin_family = AF_INET;
	tcpServerAddr.sin_addr.s_addr = inet_addr(SERVER_IP);
	tcpServerAddr.sin_len = sizeof(tcpServerAddr);
	tcpServerAddr.sin_port = htons(SERVER_PORT);
	while (1) {
		sock_fd = socket(AF_INET, SOCK_STREAM, 0);
		if (sock_fd == -1) {
			printf("ESP8266 TCP client task > socket error\r\n");
			close(sock_fd);
			vTaskDelay(1000 / portTICK_RATE_MS);
			continue;
		}
		printf("ESP8266 TCP client task > create socket: %d\r\n", sock_fd);
		
		ret = connect(sock_fd, (struct sockaddr* )&tcpServerAddr,
		sizeof(struct sockaddr));
		if (ret != 0) {
			printf("ESP8266 TCP client task > connect fail!\r\n");
			close(sock_fd);
			vTaskDelay(1000 / portTICK_RATE_MS);
			continue;
		}
		printf("ESP8266 TCP client task > connect ok!\r\n");
		
		while ((recbytes = read(sock_fd, recv_buf, TCP_DATA_LEN)) > 0) {
			recv_buf[recbytes] = 0;
			printf("ESP8266 TCP client task > recv data %d bytes!\r\n%s\r\n",
			recbytes,recv_buf);
			write(sock_fd, recv_buf, strlen(recv_buf) + 1);
			vTaskDelay(1000 / portTICK_RATE_MS);
		}
		vTaskDelay(1000 / portTICK_RATE_MS);
		recv_buf[0] = 0;
		if (recbytes <= 0) {
			printf("ESP8266 TCP client task > read data fail!\n");
			close(sock_fd);
		}
		vTaskDelay(10000 / portTICK_RATE_MS);
	}
}

void user_init(void) {
	//setup a sane baudrate
	UART_ConfigTypeDef uart_config;
	uart_config.baud_rate         = BIT_RATE_115200;
	uart_config.data_bits         = UART_WordLength_8b;
	uart_config.parity            = USART_Parity_None;
	uart_config.stop_bits         = USART_StopBits_1;
	uart_config.flow_ctrl         = USART_HardwareFlowControl_None;
	uart_config.UART_RxFlowThresh = 120;
	uart_config.UART_InverseMask  = UART_None_Inverse;
	UART_ParamConfig(UART0, &uart_config);
	
	struct station_config cfg;
	strcpy((char *)cfg.ssid, "your-wifi-ssid");
	strcpy((char *)cfg.password, "your-wifi-key");
	
	wifi_set_opmode(STATION_MODE);
	wifi_station_set_config(&cfg);
	wifi_station_set_auto_connect(1);
	
	vTaskDelay(3000 / portTICK_RATE_MS); //give wifi chance to connect
	
	xTaskCreate(tcpc_task, (signed char *)"tcpc_task", 512, NULL, 1, NULL);
}
```

suda-morris, 2015-12-29:
Thanks a lot. I have solved the problem. Because of the firewall of Win7, the TCP Server program can't get any connection from outside. When I trust it, the it works well. Thanks you very much.

closed 2015-12-29 by suda-morris

Eszartek, 2015-12-29:
Super! I'm glad to hear you solved the issue and shared the solution. I too tend to get hung up (sometimes for days) on simple issues. I find having a fully working example speeds up the troubleshooting.