espressif/ESP8266_RTOS_SDK #51: Missing support for vprintf

nitrotm, 2016-04-09:
I would like to use vprintf but it is not properly redirected to UART like printf. The problem is that I have a va_list that I'd like to pass to vprintf and there is no easy solution without that function. The function is in libcirom.a but nothing is printed out to the hook installed by os_install_putc1().

I tried to map the call to printf using __builtin_apply() but it crashes at runtime. I am not sure why.

Could you provide support for vprintf in the rtos sdk?

If you are using newlib, it should be easy to redirect all stdin/stdout/stderr using hooks similar to os_install_putc1().

valkuc, 2016-04-09:
You can use this implementation of printf with var arg support. I'm widely use it in my ESP8266 projects.

[snprintf.h.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/211313/snprintf.h.txt)
[snprintf.c.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/211314/snprintf.c.txt)

GustavoIO, 2016-05-24:
Did you used it with RTOS SDK?

I'm getting duplicate definition for vasprintf, ...

Native support like printf would be better.

BLLuis, 2016-12-08:
Seems like vprintf is inside libcirom.a at least in SDK V1.5
vprintf.o
vprintf.c
But not defined in any header, I try without success to use defining a signature for the function but the linker doesn't find the function at compiling time,

closed 2018-09-18 by FayeY