espressif/ESP8266_RTOS_SDK #54: duplicate implementations in libmain.a and libcirom.a

me-no-dev, 2016-04-17:
So far I have stumbled on 3 duplicate implementations in both libraries and opted to remove them from libcircom in order to compile.
```
lib_a-printf.o
lib_a-putchar.o
lib_a-puts.o
```

Wazalogic, 2016-07-22:
I have a similar issue with 'about'. Also it cannot find _kill and _getpid. How can I fix this problem?

-- output --

d:/Programming/ESP8266/ESP8266_RTOS_SDK/lib\libmain.a(routines.o): In function `abort':(.text+0x70): multiple definition of `abort'
d:/Programming/ESP8266/ESP8266_RTOS_SDK/lib\libcirom.a(lib_a-abort.o):/home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdlib/abort.c:55: first defined here
d:/Programming/ESP8266/ESP8266_RTOS_SDK/lib\libmain.a(routines.o): In function `__cxa_deleted_virtual':(.text+0x5): dangerous relocation: call0: call target out of range: abortd:/Programming/ESP8266/ESP8266_RTOS_SDK/lib\libmain.a(routines.o): In function `
__cxa_pure_virtual':(.text+0xd): dangerous relocation: call0: call target out of range: abort
d:/Programming/ESP8266/ESP8266_RTOS_SDK/lib\libcirom.a(lib_a-signal.o):(.irom0.literal+0x0): undefined reference to `_getpid_r'
d:/Programming/ESP8266/ESP8266_RTOS_SDK/lib\libcirom.a(lib_a-signal.o):(.irom0.literal+0x4): undefined reference to `_kill_r'
d:/Programming/ESP8266/ESP8266_RTOS_SDK/lib\libcirom.a(lib_a-signal.o): In function `_raise_r':
/home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/signal/signal.c:162: undefined reference to `_getpid_r'
/home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/signal/signal.c:163: undefined reference to `_kill_r'
collect2.exe: error: ld returned 1 exit status
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1

yanbe, 2017-06-06:
I could fix the problem by following commands. It back ups libcirom.a and excludes duplicate symbols from libcirom.a by using objcopy.
```sh
$ cp libcirom.a libcirom.a.save
$ xtensa-lx106-elf-objcopy.exe -N printf -N putchar -N puts libcirom.a
```

closed 2018-09-18 by FayeY