espressif/ESP8266_RTOS_SDK #68: Which IDE I preferred to write my freeRTOS code for ESP12E?

sarwadenj, 2016-09-29:
Hi, 
 I am very new and familiar with ESP8266 programming in arduino IDE only.
I seen freeRTOS is supported to ESP 12E. So which IDE i used to write,compile and upload the freeRTOS code into ESP12E.

So please tell me the steps to compile RTOS sdk based program and how to upload the program to the ESP8266? I am very new and familiar with ESP8266 programming in arduino IDE only.
And which IDE i use for writing.

Eszartek, 2016-09-30:
There is no IDE per se. I use a plain text editor (Geany) to edit c code. The SDK includes sample projects that compile simply from a command line shell script. Practice with an example, then move on to copying/editing one to suit your needs. It may not be flashy, but It does work well (I use Linux).

sarwadenj, 2016-10-02:
@Eszartek.... Thanks for suggestion. currently I'm thinking to use Eclipse for freeRTOS ON ESP12E is it better choice.

ivanroberto, 2016-10-02:
Hi dude, for me the fast and easy way was this :
http://www.esp8266.com/viewtopic.php?f=9&t=820&sid=fe014276db94a1147bac163cfde336d1

I think that will match with your needs.

sarwadenj, 2016-10-08:
Hi, @Eszartek , @ivanroberto thanks for helping me out.
i want to attach MPU6050 sensor(gyroscope,+accelerometer) i got it's https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050 
library files are written in C++ and i'm writing code in C.
so can i use these library ? please give me suggestion.

ivanroberto, 2016-10-08:
Yes you can,  but check out if there is no example ready in samples included. 
Good luck