espressif/ESP8266_RTOS_SDK #72: can  someone please take a look at the  README.md

SoniaZotz, 2016-10-20:
The steps outlined in  README.md  don't make any sense: 

**Modify gen_misc.sh or gen_misc.bat: For Linux:**

$export SDK_PATH=~/ESP8266_RTOS_SDK  << where is this directory ?
$export BIN_PATH=~/ESP8266_BIN               << where is this directory ?

how does one build the examples with this  limited information?

a worked and build for all of the examples would be a great help

coraminformatica, 2016-10-20:
Hello,

That folder are located where you installed the SDK.
Follow the readme of this repository [https://github.com/pfalcon/esp-open-sdk](url)
That readme help you to setup the compiling toolchain (Compiler+SDK)

Billybangleballs, 2016-12-17:
The README.md is useless. It refers to non-existent files and is of no use to anyone wanting to compile the sdk.

JBKingdon, 2016-12-17:
> The README.md is useless. It refers to non-existent files and is of no use to anyone wanting to compile the sdk.
Agreed that the documentation is not a strong point, especially if you want to install on Linux instead of windows as described in the https://espressif.com/en/support/explore/get-started/esp8266/getting-started-guide. You'll need to pull together information from quite a few different places, and then be both persistent and a bit creative. Stick with it though - the RTOS sdk is quite nice to work with and seems to be fairly stable (so far sscanf is the only major bug I've run into, although I'm pretty new to this too).

yingshancool, 2017-02-07:
hi，every body, we can use one key environment, 

ide install：`http://wiki.ai-thinker.com/doku.php/en/utils/ai_esp_ide_build`
use ide：`http://wiki.ai-thinker.com/doku.php/en/utils/ai_esp_ide_use`
b flash download：`http://wiki.ai-thinker.com/doku.php/en/utils/esp_bin_download`

impotent see ESP8266_RTOS_SDK step，step by step。

closed 2018-09-18 by FayeY