espressif/ESP8266_RTOS_SDK #74: Which address for irom0text.bin?

apmascalabs, 2016-11-21:
Hi, I have compiled the sdk and try the project_template example.  at the end return me to flash irom0text.bin at 0x20000.
Readme file recommend to flash at 0x40000. Which correct?
Other question,please : I'm using a nodeMCU board but after flash I receive only garbage on uart 0 or 1. Trying other baud rate but nothing change.
Some hint or link to understand ?
Thanks

HKRS, 2017-01-18:
Same Problem. I done irom0text.bin at 0X20000. Receiving garbage only.

@apmascalabs  did you got solution for this?

gustavomassa, 2017-01-26:
I'm having the same problem, I flash irom0text.bin at 0x20000, but it wont boot, just receiving garbage...

apmascalabs, 2017-03-09:
I know that is not solved , but I tried to build with different configuration, no more irom0text.bin but with user1.bin and bootloader 1.5 . Now it work .
But I don't consider closed this issue.
I will try again

gustavomassa, 2017-03-09:
@apmascalabs 

The right address for RTOS SDK version 1.5 is 0x20000. I've been using it for 2 months without any problems, the reboot problem that I was having was related with the instability of the reset button.

apmascalabs, 2017-03-09:
Ok, but you are talking about the no boot version? When you build which parameter are you using?

gustavomassa, 2017-04-29:
@apmascalabs 
Sorry for the late response, yes, I'm using no boot version, here is my setup:
```
Boot = none
App = 0
SPI_SPEED = 40
SPI_MODE = QIO
SPI_SIZE_MAP = 0 (512KB)

Flash adress setup(512kb):
eagle.flash.bin -> 0x00000
eagle.irom0text.bin -> 0x20000
esp_init_data_default.bin -> 0x7C000
blank.bin -> 0x7E000
```

myco, 2018-04-18:
I have same issue with my Wemos d1 mini. I Tried to write to:
0x00000 eagle.flash.bin
0x10000 eagle.irom0text.bin
0x7c000 esp_init_data_default.bin
0x7e000 blank.bin

and also:
0x00000 eagle.flash.bin
0x10000 eagle.irom0text.bin
0xFC000 esp_init_data_default.bin
0xFE000 blank.bin

and also:
0x00000 eagle.flash.bin 
0x20000 eagle.irom0text.bin 
0x7c000 esp_init_data_default.bin 
0x7E000 blank.bin

Always get some mess on serial port.
esptool.py v2.3.1

donghengqaz, 2018-09-18:
Hi Gentlemen, you may use the newer version(v3.0 or master), it is more easily to use and support more functions.

closed 2018-09-20 by FayeY