espressif/ESP8266_RTOS_SDK #75: Unable to compile driver_lib

densha500, 2016-11-25:
I have just tried to compile the driver lib in the new RTOS v 1.5.0.  

cd ~/ESP8266_RTOS_SDK/driver_lib
./make_lib.sh driver

First Error
In file included from i2c_master.c:13:0:
i2c_master.c: In function 'i2c_master_getDC':
../include/gpio.h:193:66: error: 'BIT0' undeclared (first use in this function)
 #define GPIO_INPUT_GET(gpio_no)     ((gpio_input_get()>>gpio_no)&BIT0)

Second Error
In file included from ../include/i2c_master.h:4:0,
                 from i2c_master.c:15:
i2c_master.c: In function 'i2c_master_gpio_init':
/home/densha/ESP8266_RTOS_SDK/include/espressif/esp8266/pin_mux_register.h:149:47: error: 'BIT2' undeclared (first use in this function)
         SET_PERI_REG_MASK(PIN_NAME, (((FUNC & BIT2) << 2) | (FUNC & 0x3)) << PERIPHS_IO_MUX_FUNC_S);

gustavomassa, 2017-04-27:
I'm facing the same problem.
I had to copy all the necessary driver files to my project.

peacockd, 2017-05-07:
I'm facing the same problem.  What files did you copy?

FayeY, 2018-01-05:
Hi, this issue should have been solved in the latest SDK, if it is not please feel free to reopen it.

closed 2018-01-05 by FayeY