espressif/ESP8266_RTOS_SDK #76: smartconfig_start alway crash

fwengineer, 2016-12-15:
[15:24:08.982] reset reason: 3
[15:24:08.982] epc1=0x401004e8, 
[15:24:08.982] epc2=0x00000000, 
[15:24:08.982] epc3=0x40100bdf,
[15:24:08.982] excvaddr=0x00000000, 
[15:24:08.982] depc=0x00000000
[15:24:08.982] SDK version:1.5.0-dev(950076a)

.S:
401004a8 <vPortExitCritical>:
401004a8:	ff2d21        	l32r	a2, 4010015c <xPortWantedSizeAlign+0x20>
401004ab:	000222        	l8ui	a2, a2, 0
401004ae:	62fc      	bnez.n	a2, 401004e8 <vPortExitCritical+0x40>
401004b0:	ffef21        	l32r	a2, 4010046c <xPortSysTickHandle+0x1c>
401004b3:	0020c0        	memw
401004b6:	0238      	l32i.n	a3, a2, 0
401004b8:	139c      	beqz.n	a3, 401004cd <vPortExitCritical+0x25>
401004ba:	0020c0        	memw
401004bd:	0238      	l32i.n	a3, a2, 0
401004bf:	330b      	addi.n	a3, a3, -1
401004c1:	0020c0        	memw
401004c4:	0239      	s32i.n	a3, a2, 0
401004c6:	0020c0        	memw
401004c9:	0238      	l32i.n	a3, a2, 0
401004cb:	93dc      	bnez.n	a3, 401004e8 <vPortExitCritical+0x40>
401004cd:	ffe521        	l32r	a2, 40100464 <xPortSysTickHandle+0x14>
401004d0:	000242        	l8ui	a4, a2, 0
401004d3:	744040        	extui	a4, a4, 0, 8
401004d6:	0e1466        	bnei	a4, 1, 401004e8 <vPortExitCritical+0x40>
401004d9:	0020c0        	memw
401004dc:	004232        	s8i	a3, a2, 0
401004df:	ffe221        	l32r	a2, 40100468 <xPortSysTickHandle+0x18>
401004e2:	002222        	l32i	a2, a2, 0
401004e5:	13e620        	wsr.ps	a2
401004e8:	f00d      	ret.n

yinhongxing, 2016-12-20:
我也遇到类似的问题，但我的模块是ESP8266，换回原来的配网库就没问题了，不知道是不是新库有问题

tzxespressio, 2017-03-22:
Hi sir,could you upload your project which can reproduce the issue?

FayeY, 2017-05-04:
Hi fwengineer, is this problem solved? Can we close this issue?

yinhongxing, 2017-05-04:
我没有用最新的，你关了吧

closed 2017-05-04 by FayeY