espressif/ESP8266_RTOS_SDK #77: mbedtls gives error MBEDTLS_ERR_X509_CERT_VERIFY_FAILED

muboldi, 2016-12-20:
Hi,

I'm using RTOS_SDK 1.5, trying to connect to AWS Iot.

At first i have a certificate with eliptic curve. But i get the error -15488 MBEDTLS_ERR_PK_UNKNOWN_PK_ALG. The mbedTLS documentation (https://tls.mbed.org/api/pk_8h.html#a877b66447bfa2cd49c1b99f1dc887568) describes the error as "Key algorithm is unsupported (only RSA and EC are supported)". If EC is supported in mbedTLS, why i'm getting this error? The support to EC was removed for the RTOS_SDK?

Then i create a new certificate with RSA, and i'm getting the following error in the handshake -9984 MBEDTLS_ERR_X509_CERT_VERIFY_FAILED. I use the certificate in esp-open-rtos by SuperHouse (https://github.com/SuperHouse) and it works so is not the certificate.

Here is the code to establish the connection, i take it from the aws_iot example of SuperHouse, it is similar to the Mbedtls_demo example of Espressif:

```
#include <esp_common.h>
#include <freertos/portmacro.h>
#include <lwip/sockets.h>
#include <ipv4/lwip/inet.h>
#include <lwip/netdb.h>
#include <lwip/sys.h>
#include <string.h>

// this must be ahead of any mbedtls header files so the local mbedtls/config.h can be properly referenced
#include "ssl_connection.h"

#define SSL_READ_TIMEOUT_MS           2000

const char *pers = "esp-tls";

static int handle_error(int err) {

#ifdef MBEDTLS_ERROR_C
    char error_buf[100];

    mbedtls_strerror(err, error_buf, 100);
    printf("%s\n", error_buf);
#endif
    printf("Error: %d\n", err);
    return err;
}

void ssl_init(SSLConnection* conn) {
    /*
     * Initialize the RNG and the session data
     */
    mbedtls_net_init(&conn->net_ctx);
    mbedtls_ssl_init(&conn->ssl_ctx);
    mbedtls_ssl_config_init(&conn->ssl_conf);

    mbedtls_x509_crt_init(&conn->ca_cert);
    mbedtls_x509_crt_init(&conn->client_cert);
    mbedtls_pk_init(&conn->client_key);

    mbedtls_ctr_drbg_init(&conn->drbg_ctx);
    mbedtls_entropy_init(&conn->entropy_ctx);

}

int ssl_connect(SSLConnection* conn, const char* host, int port) {
    int ret;
    char buffer[8];

    ret = mbedtls_ctr_drbg_seed(&conn->drbg_ctx, mbedtls_entropy_func,
            &conn->entropy_ctx, (const unsigned char *) pers, strlen(pers));
    if (ret < 0) {
        return -1;
    }

    ret = mbedtls_x509_crt_parse(&conn->ca_cert,
            (const unsigned char *) conn->ca_cert_str,
            strlen(conn->ca_cert_str) + 1);
    if (ret < 0) {
        return handle_error(ret);
    }

    ret = mbedtls_x509_crt_parse(&conn->client_cert,
            (const unsigned char *) conn->client_cert_str,
            strlen(conn->client_cert_str) + 1);
    if (ret < 0) {
        return handle_error(ret);
    }

    ret = mbedtls_pk_parse_key(&conn->client_key,
            (const unsigned char *) conn->client_key_str,
            strlen(conn->client_key_str) + 1, NULL, 0);
    if (ret != 0) {
        return handle_error(ret);
    }

    snprintf(buffer, sizeof(buffer), "%d", port);
    ret = mbedtls_net_connect(&conn->net_ctx, host, buffer, MBEDTLS_NET_PROTO_TCP);
    if (ret != 0) {
        return handle_error(ret);
    }

    ret = mbedtls_ssl_config_defaults(&conn->ssl_conf, MBEDTLS_SSL_IS_CLIENT,
            MBEDTLS_SSL_TRANSPORT_STREAM, MBEDTLS_SSL_PRESET_DEFAULT);
    if (ret != 0) {
        return handle_error(ret);
    }

    mbedtls_ssl_conf_authmode(&conn->ssl_conf, MBEDTLS_SSL_VERIFY_REQUIRED);
    mbedtls_ssl_conf_rng(&conn->ssl_conf, mbedtls_ctr_drbg_random, &conn->drbg_ctx);
    mbedtls_ssl_conf_read_timeout(&conn->ssl_conf, SSL_READ_TIMEOUT_MS);
    mbedtls_ssl_conf_ca_chain(&conn->ssl_conf, &conn->ca_cert, NULL);

    ret = mbedtls_ssl_conf_own_cert(&conn->ssl_conf, &conn->client_cert, &conn->client_key);
    if (ret != 0) {
        return handle_error(ret);
    }

    ret = mbedtls_ssl_setup(&conn->ssl_ctx, &conn->ssl_conf);
    if (ret != 0) {
        return handle_error(ret);
    }

    ret = mbedtls_ssl_set_hostname(&conn->ssl_ctx, host);
    if (ret != 0) {
        return handle_error(ret);
    }

    mbedtls_ssl_set_bio(&conn->ssl_ctx, &conn->net_ctx, mbedtls_net_send, NULL, mbedtls_net_recv_timeout);
    while ((ret = mbedtls_ssl_handshake(&conn->ssl_ctx)) != 0) {
        if (ret != MBEDTLS_ERR_SSL_WANT_READ
                && ret != MBEDTLS_ERR_SSL_WANT_WRITE) {
            if (ret == MBEDTLS_ERR_X509_CERT_VERIFY_FAILED) {
                return handle_error(ret);
            }
        }
        handle_error(ret);

        vTaskDelay(5000 / portTICK_RATE_MS);
    }

    mbedtls_ssl_get_record_expansion(&conn->ssl_ctx);
    ret = mbedtls_ssl_get_verify_result(&conn->ssl_ctx);
    if (ret != 0) {
        return handle_error(ret);
    }

    return ret;
}
```

In this post https://forums.aws.amazon.com/thread.jspa?threadID=219255 suggest to check the root CA certificate to validate the AWS IoT server certificate. I alredy check it.

Any idea or suggestion of what could be happen?

Thanks.

gustavomassa, 2017-02-05:
@muboldi Did you get any response?

muboldi, 2017-02-06:
@gustavomassa No, i don't get any response.

Do you have some solution or you are getting the same errors?

gustavomassa, 2017-02-06:
@muboldi  No, I think I'm going to use the open-rtos-sdk...
I did no try to use that example with the official SDK yet. I find some example on how to use the mbedtls library: 
http://iot-bits.com/articles/compiling-esp8266-mbed-tls-ssl-connection/

Are you using the same mbedtls library?

muboldi, 2017-02-07:
@gustavomassa Yes, i'm using the same library. The RTOS_SDK and open-rtos-sdk use mbedtls.

I saw that example, is the example of espressif. It works but i wasn't able to change the certificates. I mean, i change the certificates but i got the error of the post. Those certificates are fine because i use them in open-rtos-sdk.

gustavomassa, 2017-02-08:
@muboldi I got the open-rtos-sdk example and changed it to run on RTOS_SDK, but I'm having an issue to compile it with the libmbedtls.a:
`sdk/ESP8266_RTOS_SDK/lib/libmbedtls.a(ssl_tls.o):(.text.mbedtls_ssl_fetch_input+0x4): undefined reference to max_content_len`

I think we may have the same issue.. We need to re-build the mbedtls lib using the config file from open-rtos-sdk:
`#ifndef MBEDTLS_CONFIG_H
#define MBEDTLS_CONFIG_H

/* System support */
#define MBEDTLS_HAVE_ASM
#define MBEDTLS_DEPRECATED_WARNING
#define MBEDTLS_ENTROPY_HARDWARE_ALT
#define MBEDTLS_NO_PLATFORM_ENTROPY

/* mbed TLS feature support */
#define MBEDTLS_ECP_DP_SECP256R1_ENABLED
#define MBEDTLS_ECP_DP_SECP384R1_ENABLED
#define MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED
#define MBEDTLS_SSL_PROTO_TLS1_2

/* Debug support (optional) */
// #define MBEDTLS_ERROR_C
// #define MBEDTLS_DEBUG_C

/* mbed TLS modules */
#define MBEDTLS_AES_C
#define MBEDTLS_ASN1_PARSE_C
#define MBEDTLS_ASN1_WRITE_C
#define MBEDTLS_BIGNUM_C
#define MBEDTLS_CIPHER_C
#define MBEDTLS_CTR_DRBG_C
#define MBEDTLS_ECDH_C
#define MBEDTLS_ECDSA_C
#define MBEDTLS_ECP_C
#define MBEDTLS_ENTROPY_C
#define MBEDTLS_GCM_C
#define MBEDTLS_MD_C
#define MBEDTLS_NET_C
#define MBEDTLS_OID_C
#define MBEDTLS_PK_C
#define MBEDTLS_PK_PARSE_C
#define MBEDTLS_SHA256_C
#define MBEDTLS_SHA512_C
#define MBEDTLS_SSL_CLI_C
#define MBEDTLS_SSL_SRV_C
#define MBEDTLS_SSL_TLS_C
#define MBEDTLS_X509_CRT_PARSE_C
#define MBEDTLS_X509_USE_C

/* For verify RSA based root CA certificate (optional if root CA cert is signed using ECC) */
#define MBEDTLS_RSA_C
#define MBEDTLS_SHA1_C
#define MBEDTLS_PKCS1_V15
#define MBEDTLS_PKCS1_V21

/* For test certificates */
#define MBEDTLS_BASE64_C
#define MBEDTLS_CERTS_C
#define MBEDTLS_PEM_PARSE_C

/* Save RAM at the expense of ROM */
#define MBEDTLS_AES_ROM_TABLES

/* Save RAM by adjusting to our exact needs */
#define MBEDTLS_ECP_MAX_BITS   384
#define MBEDTLS_MPI_MAX_SIZE   256 // 2048 bits

/* Save RAM at the expense of speed, see ecp.h */
#define MBEDTLS_ECP_WINDOW_SIZE        2
#define MBEDTLS_ECP_FIXED_POINT_OPTIM  0

/* Significant speed benefit at the expense of some ROM */
#define MBEDTLS_ECP_NIST_OPTIM

/*
 * You should adjust this to the exact number of sources you're using: default
 * is the "mbedtls_platform_entropy_poll" source, but you may want to add other ones.
 * Minimum is 2 for the entropy test suite.
 */
#define MBEDTLS_ENTROPY_MAX_SOURCES 2

/* Save ROM and a few bytes of RAM by specifying our own ciphersuite list */
#define MBEDTLS_SSL_CIPHERSUITES                        \
    MBEDTLS_TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384

/*
 * Save RAM at the expense of interoperability: do this only if you control
 * both ends of the connection!  (See coments in "mbedtls/ssl.h".)
 * The minimum size here depends on the certificate chain used as well as the
 * typical size of records.
 */
#define MBEDTLS_SSL_MAX_CONTENT_LEN             4096

#include "mbedtls/check_config.h"

#endif /* MBEDTLS_CONFIG_H */`

I don't have the mbedtls source code that was modified for RTOS_SDK.

muboldi, 2017-02-08:
@gustavomassa I had got the same problem and i solved it defining the variable, as it is define in the official example of espressif. You can download it from here: https://github.com/espressif/esp8266-rtos-sample-code/tree/master/04Protocal/Mbedtls_demo.

`unsigned int max_content_len = 3072;`

The max_content_len variable is define in Mbedtls_demo/app/programs/programs.c. Declare it somewhere in your code and should it compile.

I copied the configuration of mbedtls library from open-rtos-sdk to RTOS_SDK and it gave me the same error (MBEDTLS_ERR_X509_CERT_VERIFY_FAILED) when it try to establish the connection whit AWS IoT. I also ported the mbedtls library of open-rtos-sdk inside my project and i was able to compile it, but again i get the same error.

But give it a try and tell me if you could make it works, maybe i did some mistake.

Good luck.

gustavomassa, 2017-02-08:
@muboldi I have good news, I'm testing here using the Mosquitto MQTT Broker instead of the AWS iot broker.
I configured the broker to use TLS 1.2 and created CA/SERVER certs, using port 8883. On the ESP8266 side I'm using just the CA Cert, then I was able to connect to broker using SSL connection and publish to topics normally. I also receive the Error: -9984, but esp is still able to connect and publish messages normally.

Now I will try to find out what is the error -9984 and why it is ocurring...

muboldi, 2017-02-08:
@gustavomassa There is something weird, you are just using the CA cert in the esp8266? Don't you flash the client cert and private key in your esp? If this is the case, are you sure that you are publishing messages over a TLS 1.2 connection and not just over MQTT?

Sorry if i'm confused but there is something that i don't get. 

Also the Error -9984 means that the certificate verification failed, so it sounds very weird that you can establish the TLS 1.2 connection.

gustavomassa, 2017-02-08:
@muboldi  I ran some tests and the anwser is yes, ESP8266 is really publishing messages over a TLS 1.2 connection.
Using FX Mqtt Client and Mqtt Lens Client, I changed the settings, removed TLS 1.2 and set toTLS 1.1, then the client(FX and Lens) wasn't able to connect to broker.
I also remove the last byte of my hard coded CA cert from the ESP8266 firmware and then it wasent able to connect to broker, receiving error -8576 and ret -128. 
When I inserted the last byte of my CA cert again, the ESP8266 was able to connect to broker over TLS 1.2 and publish messages.

I'm using Mosquitto Broker version 1.8 with SSL 1.5, here is the TLS settings on the conf file:
`# -----------------------------------------------------------------
# Certificate based SSL/TLS support
# -----------------------------------------------------------------
# The following options can be used to enable SSL/TLS support for 
# this listener. Note that the recommended port for MQTT over TLS
# is 8883, but this must be set manually.
#
# See also the mosquitto-tls man page.

# At least one of cafile or capath must be defined. They both 
# define methods of accessing the PEM encoded Certificate 
# Authority certificates that have signed your server certificate 
# and that you wish to trust.
# cafile defines the path to a file containing the CA certificates.
# capath defines a directory that will be searched for files
# containing the CA certificates. For capath to work correctly, the
# certificate files must have ".crt" as the file ending and you must run
# "c_rehash <path to capath>" each time you add/remove a certificate.
#capath

cafile /etc/mosquitto/certs/ca.crt
keyfile /etc/mosquitto/certs/server.key
certfile /etc/mosquitto/certs/server.crt

# This option defines the version of the TLS protocol to use for this listener.
# The default value allows v1.2, v1.1 and v1.0, if they are all supported by
# the version of openssl that the broker was compiled against. For openssl >=
# 1.0.1 the valid values are tlsv1.2 tlsv1.1 and tlsv1. For openssl < 1.0.1 the
# valid values are tlsv1.
tls_version tlsv1.2`

I really don't know yet why the ssl handshake is returning MBEDTLS_ERR_X509_CERT_VERIFY_FAILED, but even it is that error, the connection is still working... Maybe because the CA cert is hardcoded and we need to use the right function to load the certificate correctly.

Here is the ssl code working just with the CA cert:
`int ICACHE_FLASH_ATTR ssl_connect(SSLConnection* conn, const char* host, int port) {
    int ret;
    char buffer[8];

    ret = mbedtls_ctr_drbg_seed(&conn->drbg_ctx, mbedtls_entropy_func,
            &conn->entropy_ctx, (const unsigned char *) pers, strlen(pers));
    if (ret < 0) {
        return -1;
    }

    ret = mbedtls_x509_crt_parse(&conn->ca_cert,
            (const unsigned char *) conn->ca_cert_str,
            strlen(conn->ca_cert_str) + 1);
    if (ret < 0) {
        return handle_error(ret);
    }

//    ret = mbedtls_x509_crt_parse(&conn->client_cert,
//            (const unsigned char *) conn->client_cert_str,
//            strlen(conn->client_cert_str) + 1);
//    if (ret < 0) {
//        return handle_error(ret);
//    }
//
//    ret = mbedtls_pk_parse_key(&conn->client_key,
//            (const unsigned char *) conn->client_key_str,
//            strlen(conn->client_key_str) + 1, NULL, 0);
//    if (ret != 0) {
//        return handle_error(ret);
//    }

    snprintf(buffer, sizeof(buffer), "%d", port);
    ret = mbedtls_net_connect(&conn->net_ctx, host, buffer,
            MBEDTLS_NET_PROTO_TCP);
    if (ret != 0) {
        return handle_error(ret);
    }

    ret = mbedtls_ssl_config_defaults(&conn->ssl_conf, MBEDTLS_SSL_IS_CLIENT,
            MBEDTLS_SSL_TRANSPORT_STREAM, MBEDTLS_SSL_PRESET_DEFAULT);
    if (ret != 0) {
        return handle_error(ret);
    }

//#ifdef MBEDTLS_DEBUG_C
//    mbedtls_ssl_conf_dbg(&conn->ssl_conf, my_debug, stdout);
//    mbedtls_debug_set_threshold(5);
//#endif

    mbedtls_ssl_conf_authmode(&conn->ssl_conf, MBEDTLS_SSL_VERIFY_REQUIRED);
    mbedtls_ssl_conf_rng(&conn->ssl_conf, mbedtls_ctr_drbg_random,
            &conn->drbg_ctx);
    mbedtls_ssl_conf_read_timeout(&conn->ssl_conf, SSL_READ_TIMEOUT_MS);
    mbedtls_ssl_conf_ca_chain(&conn->ssl_conf, &conn->ca_cert, NULL);

//    ret = mbedtls_ssl_conf_own_cert(&conn->ssl_conf, &conn->client_cert,
//            &conn->client_key);
//    if (ret != 0) {
//        return handle_error(ret);
//    }

    ret = mbedtls_ssl_setup(&conn->ssl_ctx, &conn->ssl_conf);
    if (ret != 0) {
        return handle_error(ret);
    }

    ret = mbedtls_ssl_set_hostname(&conn->ssl_ctx, host);
    if (ret != 0) {
        return handle_error(ret);
    }

    mbedtls_ssl_set_bio(&conn->ssl_ctx, &conn->net_ctx, mbedtls_net_send, NULL,
            mbedtls_net_recv_timeout);

    while ((ret = mbedtls_ssl_handshake(&conn->ssl_ctx)) != 0) {
        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE) {
            if (ret == MBEDTLS_ERR_X509_CERT_VERIFY_FAILED) {
            	printf("MBEDTLS_ERR_X509_CERT_VERIFY_FAILED\r\n");
                return handle_error(ret);
            }
        }
        printf("Handle error: %d\r\n", ret);
        handle_error(ret);

        vTaskDelay(5000 / portTICK_RATE_MS);
    }

    mbedtls_ssl_get_record_expansion(&conn->ssl_ctx);
    ret = mbedtls_ssl_get_verify_result(&conn->ssl_ctx);
    if (ret != 0) {
        return handle_error(ret);
    }

    return ret;
}`

Observations: 
1 - I added this printf to make sure it was going there
`printf("MBEDTLS_ERR_X509_CERT_VERIFY_FAILED\r\n");`
2 -The vTaskDelay does not make sense, since the ssl_connection is not a task...
`printf("Handle error: %d\r\n", ret);
        handle_error(ret);

        vTaskDelay(5000 / portTICK_RATE_MS);`

I will try to load the CA cert directly from the file using the X509 load function.

PravallikaKG, 2017-03-17:
I am implementing a C program where in client and server talks to each other through socket and mbedtls is used for secure communication between client and server. While running the server and client programs at client side i am facing '-9984 - X509 - Certificate verification failed, e.g. CRL, CA or signature check failed' at 'mbedtls_ssl_handshake' call.

I referred to mbedtls/programs/ssl/ssl_server2.c and ssl_client2.c files. I have created my own ca.cert, server.cert, client.cert, server.key,client.key and ca.key files.

Following is my piece of code at server side:

```
int ret = mbedtls_x509_crt_parse_file(&cacert, ca_file);
if (ret != 0) {
mbedtls_printf( " failed\n ! mbedtls_x509_crt_parse_file returned %d\n\n", ret );
goto exit;
}

ret = mbedtls_x509_crt_parse_file(&srvcert, server_file);
if (ret != 0) {
    mbedtls_printf( " failed\n  !  mbedtls_x509_crt_parse_file returned %d\n\n", ret );
    goto exit;
}

ret = mbedtls_pk_parse_keyfile(&pkey,serverkey_file,password);
if (ret != 0) {
    mbedtls_printf( " failed\n  !  mbedtls_pk_parse_keyfile server returned %d\n\n", ret );
    goto exit;
}
..............
..............

mbedtls_ssl_conf_ca_chain( &conf, &cacert, NULL );
if( ( ret = mbedtls_ssl_conf_own_cert( &conf, &srvcert, &pkey ) ) != 0 )
{
    mbedtls_printf( " failed\n  ! mbedtls_ssl_conf_own_cert returned %d\n\n", ret );
    goto exit;
}

if( ( ret = mbedtls_ssl_setup( &ssl, &conf ) ) != 0 )
{
    mbedtls_printf( " failed\n  ! mbedtls_ssl_setup returned %d\n\n", ret );
    goto exit;
}
........................
.....................
// Performing the SSL/TLS handshake...
while( ( ret = mbedtls_ssl_handshake( &ssl ) ) != 0 )
{
    if( ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE )
    {
        mbedtls_printf( " failed\n  ! mbedtls_ssl_handshake returned %d\n\n", ret );
        goto reset;
    }
}
```

Following is the piece of my client code:

```
ret = mbedtls_x509_crt_parse_file(&cd->ca_cert, ca_cert_filename);
if (ret != 0) {
mbedtls_printf( " failed\n ! mbedtls_x509_crt_parse_file of ca returned %d\n\n", ret );
exit(ret);
}

ret = mbedtls_x509_crt_parse_file(&cd->client_cert, client_cert_filename);
if (ret != 0) {
        mbedtls_printf( " failed\n  !  mbedtls_x509_crt_parse_file of client returned %d\n\n", ret );
        exit(ret);
}

ret = mbedtls_pk_parse_keyfile(&cd->client_pkey,client_key_filename,password);
if (ret != 0) {
    mbedtls_printf( " failed\n  !  mbedtls_pk_parse_keyfile client returned %d\n\n", ret );
    exit(ret);
}

mbedtls_ssl_conf_authmode( &cd->conf, MBEDTLS_SSL_VERIFY_REQUIRED );
mbedtls_ssl_conf_ca_chain( &cd->conf, &cd->ca_cert, NULL  );

if( ( ret = mbedtls_ssl_conf_own_cert( &cd->conf, &cd->client_cert, &cd->client_pkey ) ) != 0 )
{
        mbedtls_printf( " failed\n  ! mbedtls_ssl_conf_own_cert returned %d\n\n", ret );
    exit(ret);
}
mbedtls_ssl_conf_rng( &cd->conf, mbedtls_ctr_drbg_random, &cd->ctr_drbg );
mbedtls_ssl_conf_dbg( &cd->conf, my_debug, stdout );

if( ( ret = mbedtls_ssl_setup( &cd->ssl, &cd->conf ) ) != 0 )
{
        	mbedtls_printf( " failed\n  ! mbedtls_ssl_setup returned %d\n\n", ret );
        	freeClientTLSData(ret,cd);
		exit(ret);
 }

 if( ( ret = mbedtls_ssl_set_hostname( &cd->ssl, COMMON_NAME ) ) != 0 )
 {
        	mbedtls_printf( " failed\n  ! mbedtls_ssl_set_hostname returned %d\n\n", ret );
        	freeClientTLSData(ret,cd);
		exit(ret);
 }
if( ( ret = mbedtls_ssl_setup( &cd->ssl, &cd->conf ) ) != 0 )
{
        mbedtls_printf( " failed\n  ! mbedtls_ssl_setup returned %d\n\n", ret );
        freeClientTLSData(ret,cd);
    exit(ret);
}

while( ( ret = mbedtls_ssl_handshake( &cd->ssl ) ) != 0 )
{
        if( ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE )
        {
                mbedtls_printf( " failed\n  ! mbedtls_ssl_handshake returned -0x%x\n\n", -ret );
                freeClientTLSData(ret,cd);
        exit(ret);
        }
}
```
I tried passing server common name to the api 'mbedtls_ssl_set_hostname()' and also tried passing ca cert common name. Still the issue persists. I got the common name of server or ca by double clicking those ca.cert and server.cert files.

Please let me know if i am missing anything.

Thanks in Advance.

PravallikaKG, 2017-03-19:
My issue is resolved. I am facing it because of using invalid certificates

FayeY, 2017-05-04:
Hi muboldi, is this problem still unsolved now?

closed 2018-09-18 by FayeY