espressif/ESP8266_RTOS_SDK #87: Possible to boot without WIFI enabled?

sinelabore, 2017-02-20:
Hello ,

is there a way to keep WIFI switched off for ESP8266? I did not find any kind of documentation so far. I want to start WIFI in the Arduino sketch later on once needed.

Thanks,
Peter

pipi61, 2017-02-20:
Hi!
try setAutoReconnect
[https://links2004.github.io/Arduino/d0/d5f/class_e_s_p8266_wi_fi_s_t_a_class.html#a2cd0bc5dc5cf67deae5f3293a70a21c0](url)

sinelabore, 2017-02-21:
Hello,

thanks for your answer.

I measured that it takes around 1s after power on until the arduino code gets executed and I can switch off WIFI.

Is there an earlier possibility directly after power on to switch off WIFI? Where can I add the call you suggested?

Thanks,
Peter

> 
> Hi!
> try setAutoReconnect
> https://links2004.github.io/Arduino/d0/d5f/class_e_s_p8266_wi_fi_s_t_a_class.html#a2cd0bc5dc5cf67deae5f3293a70a21c0 <x-msg://5/url>

Peter Mueller
pmueller@occsilva.de

donghengqaz, 2018-09-18:
This Function may be supported at later version, to be waiting please.