espressif/ESP8266_RTOS_SDK #88: 为什么choose bin选0可以编译成功，选1或2都编译出错？

zhongwcool, 2017-02-21:
到这一步：
STEP 2: choose bin generate(0=eagle.flash.bin+eagle.irom0text.bin, 1=user1.bin, 2=user2.bin)
选择：0，成功：
!!!
SDK_PATH: /home/alex/Desktop/ESP8266_RTOS_SDK-master
BIN_PATH: /home/alex/Desktop/ESP8266_RTOS_SDK-master/bin

No boot needed.
Generate eagle.flash.bin and eagle.irom0text.bin successully in BIN_PATH
eagle.flash.bin-------->0x00000
eagle.irom0text.bin---->0x20000
!!!

选择：1，会报错如下：
/opt/Espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
/home/alex/Desktop/ESP8266_RTOS_SDK-master/Makefile:386: recipe for target '.output/eagle/debug/image/eagle.app.v6.out' failed
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1

请问这是什么错误，应该如何解决？

编译环境: ubuntu16.10

tzxespressio, 2017-03-22:
选择的ld文件分配的Icache段的大小不满足编译的bin文件，你编译时最后一步的选择请告知

zhongwcool, 2017-03-22:
是的 我自己也发现了 选择1024什么的即可。仍然感谢。

closed 2017-03-22 by zhongwcool