espressif/ESP8266_RTOS_SDK #94: ./gen_misc.sh 'irom0.text' will not fit in region `irom0_0_seg'

marduino, 2017-04-06:
Hi,

I just finished setup the RTOS SDK environment, now I am trying to build the template application example by gen_misc.sh.

I am building "user1.bin"
my configuration is : 
`sc.sh
gen_misc.sh version 20150911

SDK_PATH:
/home/yunfei/embedded/esp8266/ESP8266_RTOS_SDK

BIN_PATH:
/home/yunfei/embedded/esp8266/ESP8266_RTOS_SDK/bin

Please check SDK_PATH & BIN_PATH, enter (Y/y) to continue:
y

Please follow below steps(1-5) to generate specific bin(s):
STEP 1: use boot_v1.2+ by default
boot mode: new

STEP 2: choose bin generate(0=eagle.flash.bin+eagle.irom0text.bin, 1=user1.bin, 2=user2.bin)
enter (0/1/2, default 0):
1
generate bin: user1.bin

STEP 3: choose spi speed(0=20MHz, 1=26.7MHz, 2=40MHz, 3=80MHz)
enter (0/1/2/3, default 2):

spi speed: 40 MHz

STEP 4: choose spi mode(0=QIO, 1=QOUT, 2=DIO, 3=DOUT)
enter (0/1/2/3, default 0):

spi mode: QIO

STEP 5: choose spi size and map
    0= 512KB( 256KB+ 256KB)
    2=1024KB( 512KB+ 512KB)
    3=2048KB( 512KB+ 512KB)
    4=4096KB( 512KB+ 512KB)
    5=2048KB(1024KB+1024KB)
    6=4096KB(1024KB+1024KB)
enter (0/2/3/4/5/6, default 0):

spi size: 512KB
spi ota map:  256KB + 256KB`

I get link error:
`xtensa-lx106-elf-gcc  -L/home/yunfei/embedded/esp8266/ESP8266_RTOS_SDK/lib -Wl,--gc-sections -nostdlib -T/home/yunfei/embedded/esp8266/ESP8266_RTOS_SDK/ld/eagle.app.v6.new.512.app1.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lcirom -lcrypto -lespconn -lespnow -lfreertos -lgcc -lhal -ljson -llwip -lmain -lmesh -lmirom -lnet80211 -lnopoll -lphy -lpp -lpwm -lsmartconfig -lspiffs -lssl -lwpa -lwps user/.output/eagle/debug/lib/libuser.a sample_lib/.output/eagle/debug/lib/libsample.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 
/home/yunfei/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
/home/yunfei/embedded/esp8266/ESP8266_RTOS_SDK/Makefile:386: recipe for target '.output/eagle/debug/image/eagle.app.v6.out' failed
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1`

marduino, 2017-04-06:
Can any please help? Thanks very much!

FayeY, 2017-04-10:
If your flash size is larger than 512KB, you can choose other options in STEP5.

marduino, 2017-04-12:
it works, thank you!

closed 2017-04-12 by marduino