espressif/ESP8266_RTOS_SDK #97: wifi_set_event_handler_cb doesn't work

nitrotm, 2017-04-17:
Since last update (~March), my handler passed to wifi_set_event_handler_cb() is never called anymore.
Is there something changed in the binary libs breaking compatibility with old releases?

FayeY, 2017-04-18:
Mine can work. Could you provide your test code?

nitrotm, 2017-04-18:
After some trial and error, I think it's linked to RTOS task priority. What's the priority of the task running the callback?
More generally, is there full documentation of what system tasks are running and at which priority? It's written user code can use priority 1 to 9 in the doc, but that's clearly wrong because it depends on the workload in the user tasks. It seems some system tasks are running at a quite low priority, such as the wifi callback.

gustavomassa, 2017-04-29:
Mine is working fine too, I've included the wifi_state_machine inside my project. I'm not working with high priority tasks yet, I will let you know if I face the same issue.

curiousmuch, 2017-05-03:
@nitrotm Thank you! I was going insane using Espressif reference code.

FayeY, 2017-05-09:
Hi, could you provide a demo which can reproduce this problem?

mmmmar, 2017-05-10:
I have the some question with you [bug in wifi_set_event_handler_cb #106](https://github.com/espressif/ESP8266_RTOS_SDK/issues/106)
do you solve it?

nitrotm, 2017-05-10:
@mmmmar After some trials, the problem is much bigger than the wifi callback. I know at least that the dynamic memory heap is also partially (rtos task memory) managed from the lowest priority in the system (idle task). That means if you have any long-running (or forever-running) task above idle priority, you are limited to a static list of tasks or you will run out of memory at some point.
The answer is basically "don't use any task without explicit vTaskDelay or other blocking mechanism". That's unfortunate and render the programming more complex, but that's manageable.

curiousmuch, 2017-05-11:
@FayeY Espressif's default example code using the handler function is a perfect example. Simply add another task with a priority >IDLE and the handler function will not be called.

curiousmuch, 2017-05-16:
http://bbs.espressif.com/viewtopic.php?f=66&t=4248&p=13174#p13174

curiousmuch, 2017-05-16:
http://bbs.espressif.com/viewtopic.php?f=66&t=4248

closed 2017-05-22 by FayeY