espressif/ESP8266_RTOS_SDK #102: system_get_time() date-time

gustavomassa, 2017-05-01:
@espressif 

I'm trying to format the current uint32 system_get_time() microseconds to date-time, here is my code:
```
#include <time.h>
#include "esp_system.h"

// Returns the local date/time formatted as 2014-03-19 11:11:52
char* getFormattedDateTime() {
	static char buffer[15];
	time_t seconds = (time_t)(system_get_time() / 1000000);
	struct tm* timeinfo = localtime(&seconds);
    strftime(buffer, sizeof(buffer), "%Y-%m-%d %H:%M:%S", timeinfo);
    return buffer;
}
```

The problem is the <time.h> library, it seems that it does not fit in the ESP8266.
`/root/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/app.out section `.text' will not fit in region `iram1_0_seg'`

Do you have any other api function to get formatted date-time?

aleksmarcelo, 2017-05-01:
Try to prefix your function with ICACHE_FLASH_ATTR, so it will be
moved to FLASH
memory segment

*   ICACHE_FLASH_ATTR char* getFormattedDateTime() { ...*

Cheers,
Marcelo Aleksandravicius

gustavomassa, 2017-05-01:
@aleksmarcelo 

I`ve already included the prefix, but it`s been generating the same error.
```
• Since the release of ESP8266_RTOS_SDK_v1.2.0, functions are stored in CACHE
area by default, and there is no need to add the macro ICACHE_FLASH_ATTR any
more. The interrupt functions can also be stored in CACHE. If users want to store
some frequently-called functions in RAM, please add IRAM_ATTR before functions’
name.
```
Thank you
Regards

aleksmarcelo, 2017-05-01:
Hi Gustavo,

Whats is the  *SPI_SIZE_MAP* you are using ?

gustavomassa, 2017-05-01:
@aleksmarcelo 

I was using SPI_SIZE_MAP 0 (4mb), I've changed to SPI_ZIE_MAP (8mb) and it is still generating the same error :(

aleksmarcelo, 2017-05-01:
Hi Gustavo,

0 (zero) is for 512KB memory map

If you are using the standard makefile try to change to 6.

What ESP are you using ?

# SPI_SIZE_MAP
# 0 : 512 KB (256 KB + 256 KB)
# 1 : 256 KB
# 2 : 1024 KB (512 KB + 512 KB)
# 3 : 2048 KB (512 KB + 512 KB)
# 4 : 4096 KB (512 KB + 512 KB)
# 5 : 2048 KB (1024 KB + 1024 KB)
# 6 : 4096 KB (1024 KB + 1024 KB)

gustavomassa, 2017-05-02:
@aleksmarcelo 

Hi Marcelo,

I'm using ESP8266-01, this is what ESP8266 Download Tool Auto Detect says about my flash:
```
flash devID:
4014h
QUAD;8Mbit
```

I don't think I can use SPI_SIZE_MAP 6, just 0,1 or 2.

aleksmarcelo, 2017-05-02:
Gustavo,

  i think you have only 1MB, so it seems be the problem.

  A lot of functions of LibC is in the flash memory.

  TDo you know the wemos modules. Good price x performance !

  Regards,

        Marcelo

closed 2017-06-12 by gustavomassa