espressif/ESP8266_RTOS_SDK #104: vApplicationIdleHook never called after wifi is started

nitrotm, 2017-05-03:
The vApplicationIdleHook is used internally by RTOS to free up memory (collect memory from dead tasks).
After starting the wifi station with dhcp, the vApplicationIdleHook is never called and so if I create / destroy a couple dozen tasks in sequence, the heap memory is quickly exhausted.

curiousmuch, 2017-05-03:
@nitrotm have you examined FreeRTOSConfig.h?
The default file has the following defines which could potentially cause you an issue: 
```
#define configIDLE_SHOULD_YIELD		1
#define INCLUDE_vTaskCleanUpResources	 0
```

Do you have another task with the priority of 0? If this is caused by the Espressif SDK, setting the IDLE task to not yield might fix your issue.

nitrotm, 2017-05-03:
Actually I link with the freertos library provided (not compiling the code in 3rd party; never managed to integrate self-compiled rtos kernel with the other espressif libraries), so changing these defines would not actually change the behaviour of the code already compiled.

I don't have any task with priority 0. And in fact, the exact same tasks don't cause any memory free issue as long as I don't start the wifi stack.

The idle-should-yield option actually insert a yield in the idle hook (internal to rtos). That seems fine to me. I am not sure vTaskCleanUpResources is used in the rtos code (maybe a legacy option).

gustavomassa, 2017-05-04:
@nitrotm 

I have the same scenario, I'm using mqtt with AWS. The wifi and iddle task are working fine.
After the last commit #100, they've changed something inside the wifi api.
To resolve my issue I've included the wifi_state_machine.c example [here](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/examples/wifi_station_machine_demo/user/wifi_state_machine.c)
Using that wifi state machine has resolved the iddle hook issue, my task's allocations/deletions are freeing memory correctly.

nitrotm, 2017-05-04:
@gustavomassa okay, thanks and good to know that you found the same problem and a solution. I'll study the difference between my code and that file. I also had the impression that this bug is new as my modules were working perfectly for about 4 months, until I upgraded the sdk.

I am well aware of the principle in FreeRTOS scheduling and priorities. It's clear that developers have to be careful when scheduling work in the tasks in order to keep a fair amount of 'idle' time available. But I don't think it's a good solution to change the preemptive policy as the wifi stack must be able to react immediately to hardware events or be exposed to rx/tx buffer overflow/underflow. Which is also important for other application, such as sampling a sensor or gpio at regular intervals. Personally I like the fact that we can make use of that.

nitrotm, 2017-05-04:
@gustavomassa I didn't find anything useful in the sample. I was not using wifi_fpm_open/close at all so I added these calls without solving the problem. BTW. in the sample you mention, there is no task at all.

gustavomassa, 2017-05-04:
@nitrotm The wifi task is created internally by the expressif lib, my problem was specifally related to the dhcp, after the last update of the SDK, they've changed the wifi api internally, so I had to manually start the dhcp before calling the connect wifi function, it was generating fatal exceptions. So adding the wifi_state_machine has corrected the problem.

nitrotm, 2017-05-05:
Alright, it turns out that it was due to the long system freeze delay when establishing the wifi connection (~ 2 [s]). This was messing up the computation of the next wakeup of one of the tasks.

closed 2017-05-05 by nitrotm