espressif/ESP8266_RTOS_SDK #105: SPIFFS issue when using spiffsimg created image

frebbles, 2017-05-06:
I have spent a while debugging and attempting to get spiffs working within the RTOS_SDK, but with the added use of creating an image with spiffsimg and uploading.

If anyone is having difficulties, it seems to be with the include/spiffs/spiffs_nucleus.h file and I have had to apply a patch that fixes/removes the _align part of the struct spiffs_page_object_ix_header as follows from line 417.

`
// object index header page header
typedef struct __attribute(( packed ))
#if SPIFFS_ALIGNED_OBJECT_INDEX_TABLES
                __attribute(( aligned(sizeof(spiffs_page_ix)) ))
#endif
{
  // common page header
  spiffs_page_header p_hdr;
#if SPIFFS_ALIGNED_OBJECT_INDEX_TABLES
  // alignment
  u8_t _align[4 - (sizeof(spiffs_page_header)&3)==0 ? 4 : (sizeof(spiffs_page_header)&3)];
#endif
  // size of object
  u32_t size;
  // type of object
  spiffs_obj_type type;
  // name of object
  u8_t name[SPIFFS_OBJ_NAME_LEN];
} spiffs_page_object_ix_header;

// object index page header
typedef struct __attribute(( packed )) {
 spiffs_page_header p_hdr;
 u8_t _align[4 - (sizeof(spiffs_page_header)&3)==0 ? 4 : (sizeof(spiffs_page_header)&3)];
} spiffs_page_object_ix;
`

Noting the added "#if SPIFFS_ALIGNED_OBJECT_INDEX_TABLES" around the align.

My question to someone who knows about this implementation... I have yet to run into a problem without it, but does the spiffs_page_object_ix require the same alignment directive... my thoughts are yes.

Thanks.
fRebbles

ustccw, 2019-01-02:
@all spiffs is not supported  currently. You could try `nvs` to save your configurations. refer to [nvs.h](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/nvs_flash/include/nvs.h)

WallaceWilliam, 2019-03-06:
in origin file 
https://github.com/pellepl/spiffs/blob/f5e26c4e933189593a71c6b82cda381a7b21e41c/src/spiffs_nucleus.h
line 501 and 517 (struct spiffs_page_object_ix_header)
u8_t _align[4 - **(**(sizeof(spiffs_page_header)&3)==0 ? 4 : (sizeof(spiffs_page_header)&3)**)**];
missed opening and closing brackets
for this reason _align is always 1 byte. it should be 3 bytes
Please write in lines 426 and 438 opening and closing brackets as in the original file (u8_t _align field)
after these fixes SPIFFS works like a clock

FayeY, 2019-05-20:
SPIFFS is added in the latest master, maybe you can have a try with it. Thanks.

closed 2019-05-20 by FayeY