espressif/ESP8266_RTOS_SDK #106: bug in wifi_set_event_handler_cb

mmmmar, 2017-05-10:
I find other also have the same issue.
if there is a user task that does not have block and delay, the function is registered in  **wifi_set_event_handler_cb** will never be callback.
I think priority of that is lower than user task, so if user task run, it has no change to callback.

curiousmuch, 2017-05-11:
I've also commented on their forum regarding this issue and have yet to receive a response.

curiousmuch, 2017-05-11:
Opps. I just checked the forum. It looks like Espressif has asked for example code. I'll provide it now.

curiousmuch, 2017-05-16:
http://bbs.espressif.com/viewtopic.php?f=66&t=4248&p=13174#p13174

curiousmuch, 2017-05-16:
Espressif is much more responsive on the forums. Please see the following:

http://bbs.espressif.com/viewtopic.php?f=66&t=4248

mmmmar, 2017-05-18:
@curiousmuch Thanks. I have saw it.

closed 2017-05-22 by FayeY