espressif/ESP8266_RTOS_SDK #107: Dhcp server memory disclosure and option parse problem

BsAtHome, 2017-05-21:
Hi,

There are several issues in the DHCP server located at "third_party/lwip/core/dhcpserver.c". Please review the following comments and suggestions to fix.

### Memory disclosure
The server sends DHCP packets out on the wire with a size based on struct dhcps_msg. However, the actual content of the packet is much smaller for almost any cases. This leads to more data on the wire (trailing data) which has not been properly initialized.

The code does:
<code>p = pbuf_alloc(PBUF_TRANSPORT, sizeof(struct dhcps_msg), PBUF_RAM);</code>
and subsequently fills that space with a (smaller) prepared packet from create_msg() and appending the options.

The data is then sent, as is:
<code>SendOffer_err_t = udp_sendto( pcb_dhcps, p, &broadcast_dhcps, DHCPS_CLIENT_PORT );</code>
Similarly for sending ACK and NAK.

The actual content-size is not adjusted. Padding is allowed (using zeros), but the extra data is not initialized. The consequence can be seen in a wireshark trace, complaining about wrong FCS because of trailing garbage. In fact, the data in the packet are left-overs from whatever is in that portion of the RAM, possibly disclosing sensitive information.

The actual size of the packet is known in the generation process (it actually keeps track of an end pointer). It is suggested that the packet pbuf is allocated with the correct size or reallocated to the actual size before sending. It should be noted that the packet size should preferably be on a word boundary, for example:
```C
...
create_msg(m);
end = add_msg_type(&m->options[4], DHCPOFFER);
end = add_offer_options(end);
end = add_end(end);

// RFC1533 says:
//   The end option marks the end of valid information in the vendor
//    field. Subsequent octets should be filled with pad options.
//
// Align the packet on word-size
// There must be enough space; probably ok for all use-cases here
while(((end - (u8_t *)m) & 3) && (end - (u8_t *)m) < sizeof(*m))
  *end++ = 0; // Padding with NOP

p = pbuf_alloc(PBUF_TRANSPORT, end - (u8_t *)m, PBUF_RAM);
...
```

### Option parser missing options
The second problem occurs when a client's answer is parsed in <code>static u8_t parse_options(u8_t *optptr, s16_t len)</code>. Only two options are parsed, 50 and 53, but have sizes 1 and 4 respectively. The code blindly copies data from the incoming packet without checking that there actually has been received data for this particular option. The options are then blindly skipped by adding the option's size-field to the pointer, missing the other fixed-size one-byte option (padding, see RFC1533).

The parser should check that:
* DHCP_OPTION_MSG_TYPE option has a length field of 1
* DHCP_OPTION_REQ_IPADDR option has a length field of 4
* the packet's end pointer encapsulates the size of the payload and the length field
* parse the padding option

A fix can be illustrated by:
```C
#define DHCP_OPTION_PADDING 0 // Missing from the header afaics
...
while(!is_dhcp_parse_end && optptr < end) {
  u16_t optadd = 0;

  switch(*optptr) {
  case DHCP_OPTION_MSG_TYPE: // 53
    if(end - optptr >= 3 && optptr[1] == 1) {
      type = optptr[2];
      optadd = 3;
    } // else optadd == 0
    break;

  case DHCP_OPTION_REQ_IPADDR: // 50
    if(end - optptr >= 6 && optptr[1] == 4) {
      ...copy IP address...
      optadd = 6;
    } // else optadd == 0
    break;

  case DHCP_OPTION_PADDING //  0x00:
    optadd = 1;
    break;

  case DHCP_OPTION_END: // 0xff
    optadd = 1;
    is_dhcp_parse_end = true;
    break;

  default:
    // All other options must be 2 bytes or more (option, size and payload)
    if(end - optptr >= 2 && end - optptr >= optptr[1] + 2) {
      optadd = optptr[1] + 2;
    } // else optadd == 0
    break;
  }

  if(!optadd) {
    // An invalid formatted option was encountered or buffer underflow
    ... return with error ...
  }

  optptr += optadd;
}

if(!is_dhcp_parse_end) {
  // We should end the parse with an end of options
  ... return with error ...
}
...
```

### Option parser call length parameter
A compounded problem exists in the call <code>s16_t ret = parse_options(&m->options[4], len);</code> (in parse_msg()), where the <code>len</code> argument is wrong. The value of len is the actual packet size received, but the parse_options() function expects the length of the options part of the packet (is simply adds the length to the start pointer). Therefore, the call should read:
```C
...
s16_t ret = parse_options(&m->options[4], len - ((u8_t *)&m->options[4] - (u8_t *)m));

// FIXME: handle error from the parse_option() call
...
```
Any possible error from parse_options() should be handled. If the packet is invalid, then there is no point of using it in the server's state-machine.

### Use of signed and unsigned types
There are numerous instances where signed and unsigned types are indiscriminately used. For example, parse_options() returns an u8_t, but is assigned to an s16_t when it is called. Many will probably not result in problems, but the chances of inadvertent bugs is huge when used in comparisons. These problems occur in parameter values, return values and locally used variables. The code should be reviewed with consistency in mind to weed out any unnecessary conversions in both size and signed-ness.

Considering the previous problems, it is suggested to do a full audit of the code to catch any other memory disclosure bugs, possible denial of service vectors, or any other significant problem.

--
Greetings Bertho

gustavomassa, 2017-05-23:
Nice job! You may post this on the Espressif forums, they are more responsive there...
http://bbs.espressif.com/viewforum.php?f=66

BsAtHome, 2017-05-23:
Well, after digging a bit further into the code, it seems as is there are some other problems. There is almost no error checking on the packets received and the used server logic results in problems when connecting some hosts (duplicate IPs, especially after server restarts). These issues are not very elegant and open for potential problems.

BTW, my comment on the length parameter being wrong must be rescinded. The parser's length parameter is actually adjusted in a previous upstream call (hardcoded subtraction). The code is a bit hard to follow now and then.

As a consequence, I'm taking a crack at rewriting the DHCP server...

closed 2018-09-18 by FayeY