espressif/ESP8266_RTOS_SDK #109: ShowCritical with the latest version

BLLuis, 2017-05-30:
I got a ShowCritical error with this version of rep as 30/05/2017 after a Scandone
using ESP8266_IOT_PLATFORM for a while with some modications, today I update ESP8266_RTOS_SDK and recompile all drivers and libs without errors, recompile main code of IOT without errors.

But after restart (and keep restarting):

mode : sta(5c:aa:dd:00:12:19) + softAP(5c:aa:dd:00:12:19)
add if0
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100

entry station mode to connect server 
bcn 0
del if1
mode : sta(5c:aa:dd:00:12:19)
scandone
ShowCritical:0

with the old version:

entry station mode to connect server 
bcn 0
del if1
mode : sta(5c:aa:dd:00:12:19)

scandone
state: 0 -> 2 (b0)
Ustate: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
pm open phy_2,type:2 0 0
cnt

gustavomassa, 2017-05-30:
same here, please take a look here:
http://bbs.espressif.com/viewtopic.php?f=66&t=4830&sid=7ca553e9f7a435e0b7bcd6aba522e5a7

BLLuis, 2017-05-30:
Hey Gustavo, thanks for your advice.

FayeY, 2017-06-01:
Hi, thanks for your reporting, we are working on this issue now. Sorry for the inconvenience.

FayeY, 2017-06-02:
Hi, this bug has been fixed, please have a try with the latest ESP8266_RTOS_SDK.

BLLuis, 2017-06-03:
Seems to be solved. Thanks.

closed 2017-06-03 by BLLuis