espressif/ESP8266_RTOS_SDK #111: TCP keep alive works in v1.4 but not in v1.5

HomeACcessoryKid, 2017-06-11:
I am having issues with the keep alive in v1.5.

My code is below. After 20 seconds a TCP keep alive sequence should be sent, but nothing happens...

I'm using  OS SDK ver: 1.5.0-dev(1aafc07) compiled @ May 17 2017 10:13:23
The very same code DOES work in OS SDK ver: 1.4.0(c599790) compiled @ Feb 26 2016 11:08:14
There every 20 seconds a TCP keep alive ack gets send and if I disable my TCPclient, it gets cleaned up with status -11.
`

        #include "esp_common.h"
        #include "espconn.h"

        struct      espconn hkcesp_conn;

        /******************************************************************************
         * FunctionName : server_recv
         * Description  : Processing the received data from the server
         * Parameters   : arg -- Additional argument to pass to the callback function
         *                pusrdata -- The received data (or NULL when the connection has been closed!)
         *                length -- The length of received data
         * Returns      : none
        *******************************************************************************/
        LOCAL void ICACHE_FLASH_ATTR
        server_recv(void *arg, char *pusrdata, unsigned short length)
        {
            struct espconn *ptrespconn = arg;
            os_printf("server rcvd a packet frm %d.%d.%d.%d:%d at %d with length %d\n", ptrespconn->proto.tcp->remote_ip[0],
                    ptrespconn->proto.tcp->remote_ip[1],ptrespconn->proto.tcp->remote_ip[2],
                    ptrespconn->proto.tcp->remote_ip[3],ptrespconn->proto.tcp->remote_port,system_get_time()/1000,length);
        }

        /******************************************************************************
         * FunctionName : server_sent
         * Description  : a packet has been sent
         * Parameters   : arg -- Additional argument to pass to the callback function
         * Returns      : none
        *******************************************************************************/
        LOCAL ICACHE_FLASH_ATTR
        void server_sent(void *arg)
        {
            struct espconn *pesp_conn = arg;

            os_printf("server sent a packet to  %d.%d.%d.%d:%d at %d\n", pesp_conn->proto.tcp->remote_ip[0],
                    pesp_conn->proto.tcp->remote_ip[1],pesp_conn->proto.tcp->remote_ip[2],
                    pesp_conn->proto.tcp->remote_ip[3],pesp_conn->proto.tcp->remote_port,system_get_time()/1000);
        }

        /******************************************************************************
         * FunctionName : server_recon
         * Description  : the connection has been err, reconnection
         * Parameters   : arg -- Additional argument to pass to the callback function
         * Returns      : none
        *******************************************************************************/
        LOCAL ICACHE_FLASH_ATTR
        void server_recon(void *arg, sint8 err)
        {
            struct espconn *pesp_conn = arg;

            os_printf("client %d.%d.%d.%d:%d disconnected with status %d\n", pesp_conn->proto.tcp->remote_ip[0],
                    pesp_conn->proto.tcp->remote_ip[1],pesp_conn->proto.tcp->remote_ip[2],
                    pesp_conn->proto.tcp->remote_ip[3],pesp_conn->proto.tcp->remote_port, err);
        }

        /******************************************************************************
         * FunctionName : server_discon
         * Description  : the connection has been disconnected
         * Parameters   : arg -- Additional argument to pass to the callback function
         * Returns      : none
        *******************************************************************************/
        LOCAL ICACHE_FLASH_ATTR
        void server_discon(void *arg)
        {
            server_recon(arg,0);
        }

        /******************************************************************************
         * FunctionName : user_accept_listen
         * Description  : server listened a connection successfully
         * Parameters   : arg -- Additional argument to pass to the callback function
         * Returns      : none
        *******************************************************************************/
        LOCAL void ICACHE_FLASH_ATTR
        server_listen(void *arg)
        {
            struct espconn *pesp_conn = arg;
            uint32_t keepalive;

            os_printf("%x  connects  from %d.%d.%d.%d:%d\n", arg, pesp_conn->proto.tcp->remote_ip[0],
                        pesp_conn->proto.tcp->remote_ip[1],pesp_conn->proto.tcp->remote_ip[2],
                        pesp_conn->proto.tcp->remote_ip[3],pesp_conn->proto.tcp->remote_port);

            espconn_regist_recvcb(pesp_conn, server_recv);
            espconn_regist_sentcb(pesp_conn, server_sent);
            espconn_regist_reconcb(pesp_conn, server_recon);
            espconn_regist_disconcb(pesp_conn, server_discon);
            espconn_set_opt(pesp_conn, ESPCONN_KEEPALIVE);
            keepalive=20;espconn_set_keepalive(pesp_conn,ESPCONN_KEEPIDLE, &keepalive);
            keepalive= 5;espconn_set_keepalive(pesp_conn,ESPCONN_KEEPINTVL,&keepalive);
            keepalive= 4;espconn_set_keepalive(pesp_conn,ESPCONN_KEEPCNT,  &keepalive);
        }

        /******************************************************************************
         * FunctionName : server_init
         * Description  : parameter initialize as a server
         * Parameters   : port -- server port
         * Returns      : none
        *******************************************************************************/
        void ICACHE_FLASH_ATTR
        server_init(uint32 port)
        {
            LOCAL esp_tcp esptcp;

            hkcesp_conn.type = ESPCONN_TCP;
            hkcesp_conn.state = ESPCONN_NONE;
            hkcesp_conn.proto.tcp = &esptcp;
            hkcesp_conn.proto.tcp->local_port = port;
            espconn_regist_connectcb(&hkcesp_conn, server_listen);

            espconn_accept(&hkcesp_conn);
            espconn_regist_time(&hkcesp_conn,7200,0); //better also use keepalive ?? 180->700seconds! 100->400s 7200->8h

        }

        /******************************************************************************
         * FunctionName : user_init
        *******************************************************************************/
        void user_init(void)
        {   
            espconn_init();
            server_init(23456);
        }

        /***********************************************************************************
         * FunctionName : user_rf_cal_sector_set
        ***********************************************************************************/
        uint32 user_rf_cal_sector_set(void) {
            extern char flashchip;
            SpiFlashChip *flash = (SpiFlashChip*)(&flashchip + 4);
            //sec_num = chip_size / flash->sector_size; but sector size is 4096 = 2^12
            uint32_t sec_num = flash->chip_size >> 12;
            return sec_num - 5;
        }
`

FayeY, 2017-06-16:
Sorry for the inconvenience. Could you use socket instead of espconn?

HomeACcessoryKid, 2017-06-16:
Thanks for your answer,

Kinda nasty to do all the rework…

Can you fix it from the espconn side? That seems just fair since it was a working feature during some releases.

Waiting for your answer,
HacK

FayeY, 2017-06-17:
Sorry for the inconvenience.

Please try to enable #define TCP_QUEUE_OOSEQ 1 in the lwipopt.h, then compile lwip by command ./make_lib.sh lwip in the directory "third_party".

HomeACcessoryKid, 2017-06-17:
Good news, this works.  
However, the comment in lwipopts.h states that it will be using more memory.
My project is already memory critical...

Can you provide a fix that does not require OutOfOrder to be enabled?
It will be good for others as well.

Thanks In Advance,
HacK

HomeACcessoryKid, 2017-07-02:
Can anyone comment if the workaround above is also supposed to be the official fix? I would expect these subject to be independent. And if so, how can it be fixed?

TIA,
HacK

FayeY, 2017-07-04:
Please have a try with these libs. [ESP8266_RTOS_Lib.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/1120809/ESP8266_RTOS_Lib.zip)

karawin, 2017-07-25:
Too many problems with this sdk on https://github.com/karawin/Ka-Radio and the workaround. I go back to SDK 1.4.2.
On low station bitrate the heap memory is down.

HomeACcessoryKid, 2017-10-29:
I have synced to todays master and am happy to see that keep alive works as before I reported it.
Thanks for the effort and case closed.

closed 2017-10-29 by HomeACcessoryKid