espressif/ESP8266_RTOS_SDK #113: mbedtls/lwip memory leak

gustavomassa, 2017-06-12:
Hello,

http://bbs.espressif.com/viewtopic.php?f=66&t=4701

User case : I'm connected to AWS IOT using mbedtls, I'm checking the internet connectivity opening a socket to google.com:80 and sending a byte every 15 seconds, also on each 4 minutes I have to close and re-open the socket, because google closes the peer after 4 minutes. 
How to test: I go to my modem interface and I disconnect the adsl client for some seconds and then re-connect it again, after some seconds the google socket fails to send/reopen resulting on an internet connectivity failed event. Thus the network controller fully destroys the mbedtls, waits for the internet (google socket) and then re-connects the mbedtls again, so between each internet disconnection/re-connection I see a memory leak of about 400 bytes. NOTE: When I disconnect the wifi interface and reconnect it again, most of the memory leak is gone. The leak is related to the mbedtls or lwip.

I'm using the mbedtls-rtos-example code:
https://github.com/espressif/esp8266-rt ... edtls_demo

This is the code used for the internet connection test:
```
#define SOCKET_HOST  ( ( const char * ) "google.com" )
#define SOCKET_PORT ( ( uint8 ) 80 )

static int socket_fd = -1;
static bool internetStatus = false;

LOCAL bool host2addr(const char *hostname , struct in_addr *in) {
    struct addrinfo hints, *servinfo, *p;
    struct sockaddr_in *h;
    memset(&hints, 0, sizeof(hints));
    hints.ai_family = AF_INET;
    hints.ai_socktype = SOCK_STREAM;
    if (lwip_getaddrinfo(hostname, 0 , &hints , &servinfo) != 0)
    {
        return false;
    }
    //loop through all the results and get the first resolve
    for (p = servinfo; p != 0; p = p->ai_next)
    {
        h = (struct sockaddr_in *)p->ai_addr;
        in->s_addr = h->sin_addr.s_addr;
    }
    lwip_freeaddrinfo(servinfo);
    return true;
}

LOCAL bool openGoogleSocket() {
    //create lwip TCP socket
    socket_fd = lwip_socket(AF_INET, SOCK_STREAM, 0);
    if( socket_fd != 0 )
    {
       IOT_ERROR("lwip_socket failed");
       lwip_close(socket_fd);
       internetStatus = false;
       return false;
    }
    //set socket force close
    lwip_force_close_set(1);
    //set TCP socket keepAlive
    uint32 opt = 0;
    opt = 1;
   if(lwip_setsockopt(socket_fd, SOL_SOCKET, SO_KEEPALIVE, &opt, sizeof(opt)) != 0) {
        IOT_ERROR("lwip_setsockop SO_KEEPALIVE failed");
      lwip_close(socket_fd);
        internetStatus = false;
        return false;
   }
    //set socket timeout
    opt = 10000;
   if(lwip_setsockopt(socket_fd, SOL_SOCKET, SO_RCVTIMEO, &opt, sizeof(opt)) != 0) {
        IOT_ERROR("lwip_setsockop SO_RCVTIMEO failed");
      lwip_close(socket_fd);
        internetStatus = false;
        return false;
   }

    //try to connect to google.com:80
    struct sockaddr_in addr;
    if (!host2addr(SOCKET_HOST, &(addr.sin_addr)))
    {
        IOT_ERROR("Invalid SOCKET_HOST: %s", SOCKET_HOST);
       lwip_close(socket_fd);
       internetStatus = false;
        return false;
    }
    addr.sin_family = AF_INET;
    addr.sin_port = htons(SOCKET_PORT);
    if( lwip_connect(socket_fd, (struct sockaddr*)&addr, sizeof(addr)) != 0 )
    {
        IOT_ERROR("lwip_connect failed");
       lwip_close(socket_fd);
        internetStatus = false;
        return false;
    }
    internetStatus = true;
    return true;
}

LOCAL void closeGoogleSocket() {
   lwip_force_close_set(1);
   lwip_close(socket_fd);
}
```

bool isInternetConnected() {
   if(!physicalStatus) {
      internetStatus = false;
      return false;
   }

   int error = 0;
   socklen_t len = sizeof (error);
   int retval = lwip_getsockopt (socket_fd, SOL_SOCKET, SO_ERROR, &error, &len);
   if (retval != 0) {
       /* there was a problem getting the error code */
      IOT_ERROR("lwip_getsockopt SO_ERROR failed");
      internetStatus = false;
      return false;
   }
   if (error != 0) {
       /* socket has a non zero error status */
      IOT_ERROR("%d", error);
      internetStatus = false;
      return false;
   }
   /* write p */
   if(lwip_write(socket_fd, "p", 1) != 1) {
      internetStatus = false;
      return false;
   }
   internetStatus = true;
   return true;
}
```

OBS: The RTOS_SDK mbedtls library "net" is using the original net.c file from ESP8266_RTOS_SDK\third_party\mbedtls\library, while the NON_OS is using the "espconn_mebdtls.c" wrapper that uses lwip behind. The RTOS_SDK mbedtls shouldn't be using the "espconn" wrapper instead of the original net file? 

Thank you
Regards

FayeY, 2018-09-25:
Could you have a try with the latest SDK, does this issue persist?

FayeY, 2018-10-22:
Closing due to lack of response, please feel free to reopen if this is still an issue with the latest SDK. Thanks.

closed 2018-10-22 by FayeY