espressif/ESP8266_RTOS_SDK #114: lwip problem on stream reception on V1.5 but not in V1.4

karawin, 2017-07-27:
I made a simple test to show the problem.
This test opens a stream with a low bitrate and receives it.
Simple no?
The problem is that time to time the stream has a problem on reception.
The test receives about 3 or 4 frames at very low speed (one per second) and stop receiving in the 4 seconds timeout:
.
.
.
Received: 1084
Received: 1440
Received: 802
Received: 638
Received: 1440
Received: 1440
Received: 1440
Received: 80

Received: -1, errno: 11

Nothing received after time_out, retry

The errno 11 is normal due to the timeout on the socket.

Testing on the sdk V1.4, this never happen even with a parallel test of both SDK. It is not a problem of the server).
 [The test source](http://karadio.karawin.fr/test/user_main.c)
[the bin to be flashed at 0x01000 with a boot file](http://karadio.karawin.fr/test/user1.4096.new.4.bin)

Thanks

karawin, 2017-07-27:
A frequent error
Some results without Received messages by with timestamp Sec:Ms:
connected with karawin2, channel 11
dhcp client start...
08.9: ip:192.168.1.253,mask:255.255.255.0,gw:192.168.1.70
09.0: Client Socket connected
01:57.3: 
Received: -1, errno: 11
01:57.4: 
Nothing received after time_out, retry
02:07.6: Client Socket connected
11:07.6: 
Received: -1, errno: 11
11:07.7: 
Nothing received after time_out, retry
11:17.9: Client Socket connected
29:13.7: 
Received: -1, errno: 11
29:13.8: 
Nothing received after time_out, retry
29:23.9: Client Socket connected
29:54.8: 
Received: -1, errno: 11
29:54.9: 
Nothing received after time_out, retry
30:05.1: Client Socket connected
32:25.7: 
Received: -1, errno: 11
32:25.8: 
Nothing received after time_out, retry
32:36.0: Client Socket connected
36:51.7: 
Received: -1, errno: 11
36:51.8: 
Nothing received after time_out, retry
37:01.9: Client Socket connected
38:29.3: 
Received: -1, errno: 11
38:29.4: 
Nothing received after time_out, retry
38:39.6: Client Socket connected
42:30.5: 
Received: -1, errno: 11
42:30.6: 
Nothing received after time_out, retry
42:40.8: Client Socket connected
50:06.5: 
Received: -1, errno: 11
50:06.6: 
Nothing received after time_out, retry
50:16.8: Client Socket connected
50:40.0: 
Received: -1, errno: 11
50:40.1: 
Nothing received after time_out, retry
50:50.3: Client Socket connected

llymaximus, 2017-11-17:
Hi, I have used your code to have some tests, the two tests are listed below:
1. Using "85.25.43.52" as the destination address, length of the data received during the life time of each socket is different, and with different networks, the test results are different;
2. Using Python write a simple server, and test with your code. The data length received each time by the client is different, compared with the length of data sent by the server, I find out that when the client's received data length is small, length of the data sent by the server is small too, and vice versa.

Therefore, network environment may have an impact on the stream reception, and length of the data received by the client may have a contact with the data sent by the server.

closed 2018-09-20 by FayeY