espressif/ESP8266_RTOS_SDK #115: Multi thread lwip bug?

knergy, 2017-07-29:
# Symptom
Opened socket never released even though calling socket_close()  function.
So free socket dry up and all tcp/ip communication stops.
( socket_close() internally defined as lwip_close_mt() )

# Fix
  Add a line shown below just behind line number 349 in ESP8266_RTOS_SDK/third_party/lowp/api/multi-threads/socket_mt.c  
`SOCK_MT_SET_SHUTDOWN(s, SOCK_MT_SHUTDOWN_OK);`

# Background
ESP8266_IOT_PLATFORM needs this SDK.
I wrote compiled software to my ESP-WROOM-02 module.
During SmartConfig through web browser, top web page implemented in ESP8266_IOT_PLATFORM/html/plug/index.html never appears.
At that time, logs out from serial port shows  
```
index  1, sockfd 4, dummy ?
index  2, sockfd 5, dummy ?
....
```

knergy, 2017-07-29:
Move to pull request.

llymaximus, 2017-12-07:
@knergy Hi, i have used single-thread task and multiple-thread task to open and close sockets, and the test result looks fine. If the problem still appears on your side, could you please provide some test code for us to debug this?

knergy, 2017-12-08:
Hi, llymaximus.
Thank you for your confirmation about my pull request.
I can send my source code which based on  ESP8266_IOT_PLATFORM and includes a little my own change.
I don't know how to send. Do you have any good idea?

llymaximus, 2017-12-11:
@knergy Hi, you can just compress your code and select them as an attachment while writing a comment for this issue.

knergy, 2017-12-18:
You can test multi thread library in ESP_IOT_PLATFORM which provided Espressif as SDK.
My source code is almost same with it.
I just started to confirm configuring my ESP-WROOM-02 and found this problem.

FayeY, 2018-09-20:
Sorry that we cannot reproduce this problem, is there any update about this issue? Does it persist with the latest SDK?

closed 2018-10-23 by FayeY