espressif/ESP8266_RTOS_SDK #117: wifi_set_opmode_current cause the Fatal exception (28)

wjzhang, 2017-08-01:
Hi.  i'm porting our ESP8266 Appilcation from NoOS SDK 2.10 to latest RTOS SDK.  but get Fatal exception 28 error form change Station mode to SoftAP+Station mode.
the   NoOS SDK 2.10 has no such problem. {our test AP is windows Hostednetwork, when get UDP connection,  test some package send/receive, then stop the hostednetwork}

here the source code:
esp_platform_startSoftAP(void)
{
    //clear udp boardcast timer
    os_timer_disarm(&udpboardcasttimer);

    //station disconnect 
    ESP_DBG("disconnect in station for SoftAP mode\r\n");  
	wifi_station_disconnect();

    //waiting get SSID/Password
    ESP_DBG("set SoftAP parameters\r\n");     
    wifi_softap_get_config(&esp8266SoftAPcfg);
    esp8266SoftAPcfg.channel = 6;
    memset(esp8266SoftAPcfg.ssid, 0, sizeof(esp8266SoftAPcfg.ssid));
    memset(esp8266SoftAPcfg.password, 0, sizeof(esp8266SoftAPcfg.password));
   
    //copy cloudID
    memcpy(esp8266SoftAPcfg.ssid, cloudSSIDId, 32);
 
    //copy password
    memcpy(esp8266SoftAPcfg.password, devPassword, 32);

    /*set SoftAP  config*/ 
    esp8266SoftAPcfg.ssid_len = 8;
    esp8266SoftAPcfg.ssid_hidden = 0;    
    esp8266SoftAPcfg.authmode = AUTH_WPA_WPA2_PSK;
    esp8266SoftAPcfg.max_connection = 1;    //only 1 Gateway

    ESP_DBG("SSID:%8s\r\n", esp8266SoftAPcfg.ssid); 
    /*set SoftAP mode*/
    ESP_DBG("set SoftAP+station mode\r\n");    
    wifi_set_opmode_current(STATIONAP_MODE);     //<== excetption
    //save current SoftAP config
    //wifi_softap_set_config_current(&esp8266SoftAPcfg);
    ESP_DBG("set SoftAP current\r\n");  
    if(wifi_softap_set_config_current(&esp8266SoftAPcfg) != true )
    {
        ESP_DBG("wifi_softap_set_config_current failed\r\n");
    }    

    /*close DHCP server*/
    ESP_DBG("stop DHCP\r\n");
    wifi_station_dhcpc_stop();
    wifi_softap_dhcps_stop();
.......
wifi_handle_event_cb(System_Event_t *evt)
{
    switch(evt->event_id)
    {
 .........................................................
        case EVENT_STAMODE_DISCONNECTED:
            ESP_DBG("disconnected AP\r\n");
            wifiConnected = false;
            if (wifi_get_opmode() == STATION_MODE)
            {
                ESP_DBG("reason:%d\r\n", evt->event_info.disconnected.reason);
                if ((evt->event_info.disconnected.reason == REASON_NO_AP_FOUND) || (evt->event_info.disconnected.reason == REASON_AUTH_FAIL))
                {
                    //clear Connected AP check 
                    checkConnectedAPTimerSet = false;
                    //Notify CoreMCU
            	    udpConnected = false;
        		    meshtalkdevice_ondisconnect();
                    //start AP mode
                    esp_platform_startSoftAP();          
                }
                else
                {
                    //re-connect
                    //wifi_station_connect();
                }
            }
            break;

here the exception log:
 2017/08/01 08:44:34:351    set Station mode
 2017/08/01 08:44:34:367    bcn 0
 2017/08/01 08:44:34:367    del if1
 2017/08/01 08:44:34:367    mode : sta(18:fe:34:f7:f4:1a)
 2017/08/01 08:44:34:367    set station config: SSID=Mesh_dmycn3ps password=bNEMrjMqgEG9PqXARAAfvAwok6UwnBaR
 2017/08/01 08:44:34:377    start DHCP
 2017/08/01 08:44:34:377    start connecting...
 2017/08/01 08:44:36:491    scandone
 2017/08/01 08:44:37:454    state: 0 -> 2 (b0)
 2017/08/01 08:44:37:454    state: 2 -> 3 (0)
 2017/08/01 08:44:37:455    state: 3 -> 5 (10)
 2017/08/01 08:44:37:455    add 0
 2017/08/01 08:44:37:455    aid 2
 2017/08/01 08:44:37:455    pm open phy_2,type:2 0 0
 2017/08/01 08:44:37:455    cnt 
 2017/08/01 08:44:37:455    
 2017/08/01 08:44:37:466    connected with Mesh_dmycn3ps, channel 11
 2017/08/01 08:44:37:466    dhcp client start...
 2017/08/01 08:44:37:466    connected AP
 2017/08/01 08:44:38:129    ip:192.168.173.102,mask:255.255.255.0,gw:192.168.173.1
 2017/08/01 08:44:38:129    got IP
 2017/08/01 08:44:38:129    start UDP client
 2017/08/01 08:44:38:129    Wifi Ready
 2017/08/01 08:44:38:129    start UDP boardcast
 2017/08/01 08:44:38:135    send UDP boardcast data
 2017/08/01 08:44:38:222    UDP connected
 2017/08/01 08:44:38:241    received CientHello
 2017/08/01 08:44:38:241    received ServerHelloACK
 2017/08/01 08:44:38:241    CoreMCU Connected
 2017/08/01 08:44:40:232     heartbeat
 2017/08/01 08:44:42:337     heartbeat
 2017/08/01 08:44:44:488     heartbeat
 2017/08/01 08:44:46:234     heartbeat
 2017/08/01 08:44:48:232     heartbeat
 2017/08/01 08:44:50:325     heartbeat
 2017/08/01 08:44:52:474     heartbeat
 2017/08/01 08:44:53:712    state: 5 -> 2 (8c0)
 2017/08/01 08:44:53:713    rm 0
 2017/08/01 08:44:53:713    pm close 7 0 0/16256920
 2017/08/01 08:44:53:713    disconnected AP
 2017/08/01 08:44:53:713    reason:8
 2017/08/01 08:44:54:704    reconnect
 2017/08/01 08:44:54:705    state: 2 -> 0 (0)
 2017/08/01 08:44:54:832    scandone
 2017/08/01 08:44:54:833    no Mesh_dmycn3ps found, reconnect after 1s
 2017/08/01 08:44:54:833    disconnected AP
 2017/08/01 08:44:54:833    reason:201
 2017/08/01 08:44:54:833    disconnect in station for SoftAP mode
 2017/08/01 08:44:54:849    set SoftAP parameters
 2017/08/01 08:44:54:849    SSID:03huq7em
 2017/08/01 08:44:54:850    set SoftAP+station mode
 2017/08/01 08:44:54:850    mode : sta(18:fe:34:f7:f4:1a) + softAP(1a:fe:34:f7:f4:1a)
 2017/08/01 08:44:54:850    add if1
 2017/08/01 08:44:54:868    bcn 100
 2017/08/01 08:44:54:868    Fatal exception (28): 
 2017/08/01 08:44:54:868    epc1=0x401000e3
 2017/08/01 08:44:54:868    epc2=0x00000000
 2017/08/01 08:44:54:868    epc3=0x40100995
 2017/08/01 08:44:54:868    epcvaddr=0x00e8b000
 2017/08/01 08:44:54:868    depc=0x00000000
 2017/08/01 08:44:54:907    þ¢@ª×Q|±T1¢)\t ò&v›©½«‡Ñ*!Ë½a§üêI*Åy1©”Ý+ÕSWTþyhõQÑþ‰Råiç{9!ù¨‡Q6-a,Å]ÔvA¬5ù*åQ‘þ       Råiç=a×w±žÅ"8!•yuz`µQ-…itxíhñ¨·—!Ù¨Y*‡uAæ´~döN|’`òÒj;¢åò§PdrdðòvQ(ÕQ jöDP‚P!1– )YéÉîA,vH›r-„t dv€öDv[…|1‰dAí‡u1En)ENø)YiIîA,r²þÄt 5TâA,P9‚)• )YiÉnA,rH›öe|5dAí‡q±E.ÅQZ…)ÙéIîA,P%Häò§P›ræp1E(…Q÷
 2017/08/01 08:44:55:167                                              éA,%Tv¡%v9AæAhAh9!Y*‡uAêIê×Qú!‘+‡{›þOSþStart esp_platform_init 07) compiled @ May 17 2017 10:13:23
 2017/08/01 08:44:55:204    start init

401000e0 <prvInsertBlockIntoUsedList>:
401000e0:	fffe41        	l32r	a4, 401000d8 <_DoubleExceptionVector+0x68>
401000e3:	0438      	l32i.n	a3, a4, 0               //<=== here the exception
401000e5:	07b327        	bgeu	a3, a2, 401000f0 <prvInsertBlockIntoUsedList+0x10>
401000e8:	438c      	beqz.n	a3, 401000f0 <prvInsertBlockIntoUsedList+0x10>
401000ea:	034d      	mov.n	a4, a3
401000ec:	fffcc6        	j	401000e3 <prvInsertBlockIntoUsedList+0x3>
401000ef:	023900        	excw
401000f2:	fffa31        	l32r	a3, 401000dc <_DoubleExceptionVector+0x6c>
401000f5:	0429      	s32i.n	a2, a4, 0
401000f7:	1248      	l32i.n	a4, a2, 4
401000f9:	0328      	l32i.n	a2, a3, 0
401000fb:	242a      	add.n	a2, a4, a2
401000fd:	0329      	s32i.n	a2, a3, 0
401000ff:	f00d      	ret.n
40100101:	000000        	ill

FayeY, 2017-08-01:
Hi,

Please try this code 
`
struct softap_config *config = (struct softap_config *) zalloc(sizeof(struct softap_config)); // initialization
    wifi_softap_get_config(config); // Get soft-AP config first.
    sprintf(config->ssid, SOFT_AP_SSID);
    sprintf(config->password, SOFT_AP_PASSWORD);
    config->authmode = AUTH_WPA_WPA2_PSK;
    config->ssid_len = 0; // or its actual SSID length
    config->max_connection = 4;
    wifi_softap_set_config(config); // Set ESP8266 soft-AP config
    free(config);
`

Thanks for your interest in ESP32!

wjzhang, 2017-08-02:
Thanks for help!
It can not work. it still cause  Fatal exception. but this time error log is difference:
 2017/08/02 17:01:44:890    disconnected AP
 2017/08/02 17:01:44:890    reason:201
 2017/08/02 17:01:44:891    disconnect in station for SoftAP mode
 2017/08/02 17:01:44:907    set SoftAP parameters
 2017/08/02 17:01:44:907    SSID:03huq7em
 2017/08/02 17:01:44:907    set SoftAP+station mode
 2017/08/02 17:01:44:907    mode : sta(18:fe:34:f7:f4:1a) + softAP(1a:fe:34:f7:f4:1a)
 2017/08/02 17:01:44:926    dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
 2017/08/02 17:01:44:926    add if1
 2017/08/02 17:01:44:926    bcn 100
 2017/08/02 17:01:44:926    set SoftAP current
 2017/08/02 17:01:44:927    bcn 0
 2017/08/02 17:01:44:927    del if1
 2017/08/02 17:01:45:870    dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
 2017/08/02 17:01:45:870    add if1
 2017/08/02 17:01:45:870    bcn 100
 2017/08/02 17:01:45:887    "àXÿ?@…#@ ˆþ"(stack_size = 0,task handle = 3fff56e8) overflow the heap_size.
 2017/08/02 17:01:45:887    Fatal exception (9): 
 2017/08/02 17:01:45:887    epc1=0x40100a3e
 2017/08/02 17:01:45:888    epc2=0x00000000
 2017/08/02 17:01:45:888    epc3=0x40221ef7
 2017/08/02 17:01:45:903    epcvaddr=0xc0aeefb6
 2017/08/02 17:01:45:903    depc=0x00000000
 2017/08/02 17:01:45:936    þ"@*×Q˜|1T±¢\t ò&v›©½«‡Qª¡Ë½a£üêÉ*Åy1)”Ù+ÕSWT~ùhõQ‘þ    Ráéç{9!y¨‡Q6-e,Å]”vA¬µy*åQÑþ‰Råiç{=a×s±žÅ"8!•yuz`µQ-…itxéhñ(·×!Ù(Y*‡uAf´~döN|’`òÒj;¢åò§PdrdðòvQ(ÕQ   jöDP‚P!1–$)YéÉîA,vH›r-„t dv€öDv[…|1‰dAí‡u1En)    ò)YiIîA,v…e|1‰dAí‡u1EnÅQ·
 2017/08/02 17:01:46:074                                                                             )YiÉnA,r²þÄt5TâA,P91Z…)ÙéIîA,ZZ…åò§P›æp1A(…
 2017/08/02 17:01:46:197                                        éA,%Tv¡%ö9AæAhAh9!Y*‡uAêIê×Qz!‘+‡{›þOS SDK vþStart esp_platform_init mpiled @ May 17 2017 10:13:23
 2017/08/02 17:01:46:235    start init
 2017/08/02 17:01:46:235    chipID=16249882
 2017/08/02 17:01:46:235    mode=0
 2017/08/02 17:01:46:235    set NULL mode
 2017/08/02 17:01:46:236    Read Boot flash layout
 2017/08/02 17:01:46:247    FOTA image store sector is = 128
 2017/08/02 17:01:46:247    system Ready
 2017/08/02 17:01:46:247    init sercurity lib
 2017/08/02 17:01:46:247    mode : null

BTW! which stack use by the wifi_set_event_handler_cb callback? is call wifi_set_event_handler_cb() task?  or WiFi event task? if WiFi event task, how to modify it stacks size?

llymaximus, 2017-11-30:
Hi, I've tried your source code with the latest SDK version, the test result looks fine and doesn't come with any exception, the attachment is my test code and test log.
[issue_117.tar.gz](https://github.com/espressif/ESP8266_RTOS_SDK/files/1516768/issue_117.tar.gz)

wjzhang, 2017-12-01:
Thanks a lot!

Best regards!
zhangwenjin

closed 2018-09-18 by FayeY