espressif/ESP8266_RTOS_SDK #118: User Ping.c Ping.h always timeout?

KISSMonX, 2017-08-11:
anyone implement ping command ? 
help me ? Thanks.

DKrepsky, 2017-08-21:
Someone removed the raw sockets receive buffer.

To fix that enable the receiver buffer in  "SDK_PATH"/include/lwip/lwipopts.h by adding the following line:
`#define DEFAULT_RAW_RECVMBOX_SIZE       6`

After that, rebuild lwip lib:
```
rm $SDK_PATH/lib/liblwip.a
cd $SDK_PATH/third_party
./make_lib.sh lwip
```
This should enable lwip_recvfrom and raw_recv.

Also, If you need the source code for a ping function I can provide it.

Regards.

KISSMonX, 2017-08-22:
@DKrepsky Although I ported the ping and can be used, but there are still problems with memory leaks.
If you can provide a code, I would be very grateful.

DKrepsky, 2017-08-27:
You can check my implementation here:
https://github.com/DKrepsky/ESP8266-Utils

KISSMonX, 2017-08-27:
@DKrepsky Thank you very much. :)

closed 2018-09-18 by FayeY