espressif/ESP8266_RTOS_SDK #121: How to save/create files with spiffs?

beckmx, 2017-09-08:
I have been trying to save a file with spiffs but the compiler complains about this error

`error: 'fs' undeclared`

I have these headers

```
#include "esp_common.h"
#include "hkc.h"
#include "gpio.h"
#include "queue.h"

#include "spiffs.h"
#include <stdlib.h>
#include "spiffs_test_params.h"

#include "fcntl.h"
#include "unistd.h"

```

Anyone knows how to use spiffs or a better way to create or save data?

ustccw, 2019-01-02:
@beckmx spiffs is not supported  currently. You could try `nvs` to save your configurations. refer to [nvs.h](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/nvs_flash/include/nvs.h)

WallaceWilliam, 2019-03-06:
in esp_spiffs.c replace
static spiffs fs; ->  spiffs fs;
in esp_spiffs.h add in top
 spiffs fs;
and patch [patch](https://github.com/espressif/ESP8266_RTOS_SDK/issues/105#issuecomment-470250562)

FayeY, 2019-05-20:
SPIFFS is added in the latest master, maybe you can have a try. Thanks.

closed 2019-05-20 by FayeY