espressif/ESP8266_RTOS_SDK #122: API system_get_flash_size_map() always return 4

wlcwjy, 2017-09-08:
Download the SDK 1.5
make BOOT=new APP=1 SPI_SPEED=26.7 SPI_MODE=DIO SPI_SIZE_MAP=3
Create bin user1.2048.new.3.bin
The starting byte for bin file already change:EA 04 02 31 04 10 40 00

make BOOT=new APP=1 SPI_SPEED=26.7 SPI_MODE=DIO SPI_SIZE_MAP=4
Create bin user1.4096.new.4.bin
The starting byte for bin file already change:EA 04 02 41 04 10 40 00

make BOOT=new APP=1 SPI_SPEED=26.7 SPI_MODE=DIO SPI_SIZE_MAP=5
Create bin /user1.2048.new.5.bin
The starting byte for bin file already change:EA 04 02 51 04 10 40 00

print log aways like that

code:TRACE_NOTICE("Userbin bin is:%d,Boot mode:%d,Cpu freq:%d,Flash map:%d,Bin addr:0x%08X\r\n",
		system_upgrade_userbin_check(),system_get_boot_mode(),system_get_cpu_freq(),
		system_get_flash_size_map(),system_get_userbin_addr());
print log:[NOTICE][user_init:215]:Userbin bin is:1,Boot mode:1,Cpu freq:80,Flash map:4,Bin addr:0x00081000

**Flash map:4**    system_get_flash_size_map() always return 4

I check the *.S and *.dump file,system_get_flash_size_map() seem contain in libmain.a,could you fix this BUG?

llymaximus, 2017-11-17:
Hi, I've tried the compile options mentioned above, and always get 4 as a result of the API system_get_flash_size_map(). Also tried with changing the flash_size option while using the esptool.py to download bin files, and here is the result:
1. esptool.py --port /dev/ttyUSB0 --baud 921600 write_flash **--flash_size 32m** 0x00000 ../bin_1_5/eagle.flash.bin 0x20000 ../bin_1_5/eagle.irom0text.bin 0x3fc000 ../bin_1_5/esp_init_data_default.bin 0x3fe000 ../bin_1_5/blank.bin 0x3ff000 ../bin_1_5/blank.bin
The result of system_get_flash_size_map() is **4**
2.  esptool.py --port /dev/ttyUSB0 --baud 921600 write_flash **--flash_size 32m-c1** 0x00000 ../bin_1_5/eagle.flash.bin 0x20000 ../bin_1_5/eagle.irom0text.bin 0x3fc000 ../bin_1_5/esp_init_data_default.bin 0x3fe000 ../bin_1_5/blank.bin 0x3ff000 ../bin_1_5/blank.bin
The result of system_get_flash_size_map() is **6**
So the result may be decided by the flash_size you choose while using the esptool.py to download.

27-301, 2018-03-16:
@wlcwjy Can you share the 2nd bootloader log?

fafactx, 2018-05-17:
I have the same problem，and thanks @llymaximus 。you help me solved the big problem。

closed 2018-06-15 by 27-301