espressif/ESP8266_RTOS_SDK #125: Version and update third_party libs

beriberikix, 2017-10-10:
Just looking at `third_party`, it is hard for me to know which version the SDK is using. However, based on what I can tell (copyright dates, included files, etc.), these dependencies are pretty old. For future updates, can you please include the version of the library in the commit message?

Now, it would be great to move to more recent versions of these dependencies for obvious stability/security/resource reasons. For reference, here are the latest version at the time of posting:

- [FreeRTOS v9.0.0](https://sourceforge.net/projects/freertos/files/FreeRTOS/)
- [cJSON v1.6.0](https://github.com/DaveGamble/cJSON/releases)
- [lwIP 2.03](http://download.savannah.nongnu.org/releases/lwip/)
- [mbed TLS 2.6.0](https://github.com/ARMmbed/mbedtls/releases)
- [noPoll 0.4.5](https://github.com/ASPLes/nopoll/releases)
- [OpenSSL 1.1.0f](https://github.com/openssl/openssl/releases)
- [SPIFFS 0.3.7](https://github.com/pellepl/spiffs/releases)
- [axTLS 2.1.4](https://sourceforge.net/projects/axtls/files/)

Understandably blindly upgrading libraries is not a good strategy, especially if doing so might introduce new bugs or consume more resources but some of these libraries could have patched issues that are critical (I have not personally confirmed.)

Thank you for considering this!

27-301, 2018-03-16:
Will do this after SDK v2.0.0 released, there are bunch of things to do, but all are in plan. As you know, it might introduce new bugs, so the steps will be such as this: FreeRTOS->lwip->mbedTLS, these are fundamental and critical.

Thank you for your advice.

27-301, 2018-06-07:
@beriberikix Most third party libraries are upgraded, you can check from master branch. And now, the SDK frame is changed to idf style, you can have a try. Thanks.

closed 2018-06-07 by 27-301