espressif/ESP8266_RTOS_SDK #127: espconn source code missing mdns functions

HomeACcessoryKid, 2017-11-01:
In august the source code of espconn has been added a third party source code
The header file refers to the mdns routines.
However, the source code files are missing these mdns functions.

Can you also indicate what is the reason of offering the espconn code of the nonos branch (is it??)

Please add the missing files to the repository and verify it all works?
Thanks

donghengqaz, 2018-09-18:
This Function may be supported at later version, to be waiting please.

FayeY, 2018-10-29:
It is recommended to use the new ESP8266 RTOS SDK V3.0 or later versions which is in ESP-IDF style.  We just provide the mdns function on GitHub recently, please have a try. Thanks.

closed 2018-10-29 by FayeY