espressif/ESP8266_RTOS_SDK #133: RTOS_SDK_1.5 espconn error

ukrsms, 2017-11-30:
Till now I used SDK v 1.3 but I decided to switch to 1.5 (current version) RTOS SDK. I downloaded SDK from here https://github.com/espressif/ESP8266_RTOS_SDK/tree/master and tried to compile it, but I got an error:

```
c:/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -Lc:/Espressif/ESP8266_RTOS_SDK/lib -Lout/build -Trom0.ld -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group, out/build/app_app.a -lgcc -lhal -lphy -lpp -lmain -lssl -lnet80211 -lespconn -lwpa -lcrypto -lfreertos -lcirom -llwip -lminic -lpwm -lsmartconfig -Wl,--end-group -Wl,-Map,foo0.map -o out/build/app_0.out
c:/Espressif/ESP8266_RTOS_SDK/lib\libespconn.a(espconn_buf.o):(.text.ringbuf_memset+0xc): undefined reference to `os_memset'
c:/Espressif/ESP8266_RTOS_SDK/lib\libespconn.a(espconn_buf.o): In function `ringbuf_memset':
(.text.ringbuf_memset+0x6a): undefined reference to `os_memset'
collect2.exe: error: ld returned 1 exit status
```
I tried to find the os_memset function but it seems it is absent in the SDK.
I've studied the issue more deeply and found that the problem is connected with espconn lib.
If I do not use any espconn functions then project compiles without errors, but if there are some espconn functions in the code the mentioned error occurs.

For example, I've tested the issue on a simple code but with the same errors:
```
#include "esp_common.h"
void upgrade_recon_cb(void *arg, sint8 errType)
{
   printf("Connection error: ");
}
void user_init(void)
{
   struct espconn *conn;
   espconn_regist_reconcb(conn, upgrade_recon_cb);
}
void user_rf_cal_sector_set(void)
{
}
```
I tried other espconn functions instead of espconn_regist_reconcb but the result was the same.
I'm confused with the issue because it is hard to believe that such an error is there in new SDK 
Does anybody use RTOS_SDK_1.5 ?

ukrsms, 2017-12-04:
I excluded _espconn lib_ from my makefile and include source code of espconn implementation provided in third_party dir of the SDK into my project. However, I found there two problem lines:
_file espconn_buf.c_
1. in function ringbuf_findchr()
`  const uint8_t* found = (const uint8_t*)memchr(start, c, n);`
here memchr is used for recursive operation and causes a compilation error (I do not know why)
This issue I solved with declaration of alternative implementation of memchr.
1. in function ringbuf_memset()
`os_memset(dst->head, c, n);`
the function os_memset is absent in the whole RTOS_SDK so I just changed it for memset()
Therefore I compiled SDK without any errors.

FayeY, 2018-10-29:
Please use the socket instead, it is a better way to create connections. It is recommended to use the new ESP8266 RTOS SDK V3.0 or later versions which is in ESP-IDF style. Sorry for the inconvenience.

closed 2018-10-29 by FayeY