espressif/ESP8266_RTOS_SDK #138: Can't deep sleep

ivanroberto, 2017-12-31:
Hi guys,
I have been in troubles trying to put my ESP8266 in Deep Sleep, using the MASTER ESP_RTOS_SDK.
If I rollback to the v1.4 branch my ESP deep sleep flawless.
I already tried to recompile all third part libs and drive libs, but no luck.

It seems to try to deep sleep, but maybe the watch dog to not let it happend.

Thanks a lot!

Here is a simple code that i made to debug this:

```
#define DEBUG
#include "esp_common.h"
#include "espressif/espconn.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "uart.h"

os_timer_t sensor_sleep_timer;

#define TIME_TO_DEEP_SLEEP	20 //Tempo em minutos

uint32_t user_rf_cal_sector_set(void){
    flash_size_map size_map = system_get_flash_size_map();
    uint32_t rf_cal_sec = 0;

    switch (size_map) {
        case FLASH_SIZE_4M_MAP_256_256:
            rf_cal_sec = 128 - 5;
            break;

        case FLASH_SIZE_8M_MAP_512_512:
            rf_cal_sec = 256 - 5;
            break;

        case FLASH_SIZE_16M_MAP_512_512:
        case FLASH_SIZE_16M_MAP_1024_1024:
            rf_cal_sec = 512 - 5;
            break;

        case FLASH_SIZE_32M_MAP_512_512:
        case FLASH_SIZE_32M_MAP_1024_1024:
            rf_cal_sec = 1024 - 5;
            break;

        default:
            rf_cal_sec = 0;
            break;
    }

    return rf_cal_sec;
}

void sleepTimeOutTask(void *pvParameters){
	vTaskDelay(20000 / portTICK_RATE_MS);	//Tempo maximo em segundos para forçar um deep sleep
	os_printf("Sleep...\n");

//	system_deep_sleep_set_option(2);
	system_deep_sleep(TIME_TO_DEEP_SLEEP*60*1000*1000);		//Sleep por X minutos
	vTaskDelay(1000 / portTICK_RATE_MS);
	vTaskDelete(NULL);
}

void wifi_task(void *pvParameters){
    struct station_config sta_config;
    struct ip_info ip_config;
    bzero(&sta_config, sizeof(struct station_config));

    SET_PERI_REG_MASK(UART_CONF0(0), UART_TXFIFO_RST); //Reset FIFO UART
    CLEAR_PERI_REG_MASK(UART_CONF0(0), UART_TXFIFO_RST);

  	sprintf(sta_config.ssid, "blabla");
  	sprintf(sta_config.password, "19811985");
    wifi_station_set_config(&sta_config);
  	os_printf("%s\n", __func__);
  	wifi_get_ip_info(STATION_IF, &ip_config);
  	os_printf("Connecting...\n");
  	while(ip_config.ip.addr == 0){
  		vTaskDelay(100 / portTICK_RATE_MS);
  		wifi_get_ip_info(STATION_IF, &ip_config);
  	}
  	os_printf("Connected\n");
  	xTaskCreate(sleepTimeOutTask, "sleepTimeOutTask", 256,	NULL, 2, NULL);
    vTaskDelete(NULL);
}

void user_init(void){
    user_rf_cal_sector_set();
    UART_ResetFifo(UART0);
    uart_init_new();
    UART_SetBaudrate(UART0,BIT_RATE_115200);

    wifi_set_opmode(STATION_MODE);
    uint8_t wifiStatus = STATION_IDLE;
#ifdef DEBUG
    os_printf("SDK version:%s\n\r", system_get_sdk_version());
#endif
    xTaskCreate(wifi_task, "wifi_task", 512, NULL, 2, NULL);
}
```

And that is the result os UART print:
```
Connecting...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with blabla, channel 1
dhcp client start...
ip:192.168.1.109,mask:255.255.255.0,gw:192.168.1.1
Connected
Sleep...
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/23790481
del if0
usl
sul 0 0
mode : null
Fatal exception (0): 
epc1=0x40224ddb
epc2=0x00000000
epc3=0x4010071e
epcvaddr=0x40107681
depc=0x00000000
útn_add=0x40224ddb
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 30312, room 16 
tail 8
chksum 0x71
load 0x3ffe8000, len 808, room 0 
tail 8
chksum 0xef
load 0x3ffe8330, len 908, room 0 
tail 12
chksum 0xc9
csum 0xc9
rll|là|lìb|ìrbbònnlnnâìbpllblrlpònàlbnâ|ììlbònnîll`nnl`nrnòll`pònàrâàbnâ|bònnîl`nnl`nrnâ`pònàrâàììbnâ|lbònnîl`nnll`nrnrl`rnrl`~rìbâblbr|lbâbrpâìlbìònînnââllìlll`nüâþÊSDK version:1.5.0-dwifi_task
Connecting...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with blabla, channel 1
dhcp client start...
ip:192.168.1.109,mask:255.255.255.0,gw:192.168.1.1
Connected
Sleep...
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/24297868
del if0
usl
sul 0 0
mode : null
Fatal exception (0): 
```

ivanroberto, 2018-01-06:
I made a test here and realized that before last commit deep sleep works. (This on 3577acc9b80c3dfb20b77b2c50661d093ed93263 Day 26 December is OK).
So I guess that recompile libs with gcc (last commit) made things not so good. 
Maybe Mr. wujiangang, who made this commit can help us???
Thanks

FayeY, 2018-01-08:
Hi, sorry for the inconvenience.
We just updated the RTOS SDK, which should solve this issue, please have a try.
Thanks.

ivanroberto, 2018-01-13:
Hi Mr. Faye Y,

It works!

Thank you so much!!!

closed 2018-01-13 by ivanroberto