espressif/ESP8266_RTOS_SDK #139: It's not possible get STATION_WRONG_PASSWORD or STATION_NO_AP_FOUND

wicked-code, 2017-12-31:
I want to notify the user (using STATIONAP_MODE mode) about wrong ssid or password, but right now it's not possible. In such case (password or ssid is incorrect) ESP automatically trying to reconnect despite the status of reconnect policy (wifi_station_set_reconnect_policy(false);). So, wifi_station_get_connect_status() never returns STATION_WRONG_PASSWORD or STATION_NO_AP_FOUND.

wicked-code, 2018-01-07:
Looks like wifi_station_get_connect_status() is not updated in low priority threads. Now my setup thread has priority 9 and wifi_station_get_connect_status() works correctly. So, I have workaround for now, but I think the problem still exists. Also I should mention that wifi_station_set_reconnect_policy(false) still ignored. Thus I have to stop connecting (wifi_station_disconnect()) in case of wrong password or wrong ssid name.

mcuwty, 2018-11-12:
I also encountered the same problem. I can get the STATION_WRONG_PASSWORD or STATION_NO_AP_FOUND in the ESP8266_NONOS_SDK via the wifi_station_get_connect_status(). But now I can't use it in ESP8266_RTOS_SDK. It seems to be undefined. And I can't find an alternative function or error code. I think this feature is very important and useful because you can notify the user that the ssid or password is wrong.
### Environment
IDF version (git rev-parse --short HEAD to get the commit id.):
9d6c560

mcuwty, 2018-11-12:
This issue seems to have been open for a long time. Is there a plan to solve this problem?

FayeY, 2018-12-28:
Hi, could you have a try with our latest master? If the problem persists, please feel free to let us know. Thanks.

closed 2019-01-09 by FayeY