espressif/ESP8266_RTOS_SDK #142: SSL/TLS : receive buffer management issue

Totalus, 2018-01-04:
I found a bug in the SSL code related to the receive buffer management.

**Bug**
`ssl_read()` function returns the -261 error code ( SSL_ERROR_INVALID_PROT_MSG ) when trying to receive a considerable ammount of data over SSL (more than 3072 bytes in a single SSL Record).

I've traced the problem up to the `basic_read()` function, [line 1328](https://github.com/espressif/ESP8266_RTOS_SDK/blob/eb91560c0c697381dbeb20d2fa8335e2952007a6/third_party/ssl/ssl/ssl_tls1.c#L1326). The condition evaluates to true and sets the error code.

**Source of the bug**
The `basic_read()` function intends to read the full SSL Record and store it to the `ssl->bm_all_data` buffer before returning. The maximum SSL Record size is 16383 bytes for TLS v1 ([source](https://www.cisco.com/c/en/us/support/docs/security-vpn/secure-socket-layer-ssl/116181-technote-product-00.html#anc5)), however the buffer is much smaller (3072 bytes).

The original axTLS library uses a big buffer (16kb+), but it was changed for the ESP implementation because of memory constraints.

**Suggestion for Solution**
The buffer management should probably be modified so it streams the data out when the SSL Record size is higher than the receive buffer size, instead of trying to read all the record before returning.

Totalus, 2018-01-09:
**Here is an update after further investigation**
We can't really stream the data out as I suggested. There is a verification digest at the end of the SSL record and you need all the data to compute it, unless it is ignored, but then the SSL implementation would probably not follow the specs and it will reduce security.

TLS has an extension that allows limiting the fragment length (refer to RFC6066), but it does not not seem to be supported by the _axTLS_ library. The modification that were made to use fragment length are not complete and prevent the use of the library for big SSL records.

**Conclusion**
_axTLS_ seems obsolete, though it is used in some examples of the IoT Platform, _mbedTLS_ is much more complete for TLS/SSL implementation. Both libraries are included in the Espressif RTOS SDK. As a suggestion for developpers, use _mbedTLS_ (library/mbedtls) rather than _axTLS_ (library/ssl).

closed 2018-01-09 by Totalus

Junhao-Espressif, 2018-01-09:
Hi @Totalus 

We have a function ```SSL_CTX_set_default_read_buffer_len``` to set the length of the buffer. 
The range is 2048 ~ 8192 bytes.

Totalus, 2018-01-09:
That is an _openssl_ library function. How many SSL library does the SDK have ?

Reducing the receive buffer length is not enough if SSL records received are too long. The SSL implementation must implement the _max_fragment_length_ extension (refer to RFC6066) of TLS v1 to ask the server to reduce the SSL record size to fit the buffer, otherwise reading will return an error (-261 for axTLS and -0x7200 for mbedTLS). You either need a control on the server (flush the data out of send buffer before it gets to a certain size) or a server that also supports the _max_fragment_length_ extension.