espressif/ESP8266_RTOS_SDK #147: ESPNow Broadcast with MAC ff:ff:ff:ff:ff:ff is not working

DieterWagner, 2018-02-19:
In the ESP32 the ESPNOW supports broadcasting using the MAC ff:ff:ff:ff:ff:ff, but with the ESP8266 the broadcast is not working.
Please add the same ESPNOW  behavior of the ESP32 to the ESP8266.

Crazor, 2018-05-29:
Sending broadcasts from the 8266 seems to work (at least the function does not return an error), but the broadcast packets are never received by other 8266's.

Junhao-Espressif, 2018-09-20:
Hi @DieterWagner @Crazor ,

In previous versions, broadcast packets was not supported.
Could you try espnow on release/v3.0 branch?

DieterWagner42, 2018-09-23:
Hi,
thanks for the new release/v3.0 branch. so I did some tests which were almost successful.
Scenarion with some issues:
ESP32 with espnew started first. (I set the magic nummer always to null)
ESP8266 with espnew from release/v3.0 starts second.
this is whats happans almoast always:
I (35) boot: ESP-IDF v3.1-dev-153-g6efd160 2nd stage bootloader
I (35) boot: compile time 11:56:46
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : QIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000f0000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x40568 (263528) map
I (145) esp_image: segment 1: paddr=0x00050580 vaddr=0x3ffe8000 size=0x00534 (  1332) load
I (145) esp_image: segment 2: paddr=0x00050abc vaddr=0x3ffe8534 size=0x0012c (   300) load
I (152) esp_image: segment 3: paddr=0x00050bf0 vaddr=0x40100000 size=0x069f0 ( 27120) load
0x40100000: _stext at ??:?

I (166) boot: Loaded app from partition at offset 0x10000
I (12) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (17) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
mode : sta(5c:cf:7f:2c:18:e7)
add if0
I (225) espnow_example: WiFi started
I (5221) espnow_example: Start sending broadcast data
I (6221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (7221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (8221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (9221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (10221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (11221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (12221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (13221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (14221) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (14226) espnow_example: Receive 0th broadcast data from: 24:0a:c4:09:8c:d8, len: 100
I (15231) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (15237) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (16231) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (16237) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (17231) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (17237) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (18231) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (18237) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
E (18659) espnow_example: Receive cb arg error
I (19231) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (20231) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (20236) espnow_example: Receive ESPNOW data too short, len:1
I (20239) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (21241) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (21245) espnow_example: Receive ESPNOW data too short, len:2
I (21249) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (22251) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (22255) espnow_example: Receive ESPNOW data too short, len:3
I (22259) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (23261) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (23265) espnow_example: Receive ESPNOW data too short, len:4
I (23269) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (24271) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (24275) espnow_example: Receive ESPNOW data too short, len:5
I (24279) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (25281) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (25285) espnow_example: Receive ESPNOW data too short, len:6
I (25289) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (26291) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (26295) espnow_example: Receive ESPNOW data too short, len:7
I (26299) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (27301) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (27306) espnow_example: Receive ESPNOW data too short, len:8
I (27309) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8
I (28311) espnow_example: send data to ff:ff:ff:ff:ff:ff
E (28315) espnow_example: Receive ESPNOW data too short, len:9
I (28319) espnow_example: Receive error data from: 24:0a:c4:09:8c:d8

When I rest the ESP8266 some time it works. It seams that the callback doesn't receive a full package.
Thanks
Dieter

closed 2018-09-26 by wujiangang

DieterWagner42, 2018-10-01:
Hi,
thanks Now it is working.
Dieter