espressif/ESP8266_RTOS_SDK #153: 请问MQTT库是否有计划完成ssl版本的开发

jayyuz, 2018-03-10:
如题，是否有计划呢？如果有，大概什么时间能够完成呢？

27-301, 2018-03-16:
Hopefully next week.

jayyuz, 2018-03-24:
A week passed，it seems that nothing has happened.

27-301, 2018-03-27:
@jayyuz commit d2a5facdca5fb599dcbdd08e072bcf0fc98fa808 adds the support of SSL, have a try.

jayyuz, 2018-03-28:
Good job! After an hour of testing, it runs very well. @27-301

closed 2018-03-29 by wujiangang

xiangjiangnan, 2018-04-28:
请问按上述工程，esp8266加载证书时异常，是存储空间受限的原因吗？错误如下
WIFI_INIT
mode : sta(5c:cf:7f:f8:4a:8d)
add if0
wifi_check_ip
STATION_IDLE
load cart.
Fatal exception (0): 
epc1=0x40279d20
epc2=0x00000000
epc3=0x402818a7
epcvaddr=0x40108d98
depc=0x00000000
rtn_add=0x40278781

Free Heap Size: 42528
Stack Point: 3fff7e10
3fff7e10: 401086c0 00000362 3fff8c33 00000000
3fff7e20: 00000001 00000000 00000000 00000000
3fff7e30: 00000000 00000000 00000000 00000000
3fff7e40: 00000000 00000000 00000000 4027da17
3fff7e50: a5a5a5a5 00000000 40107948 40108a38
3fff7e60: 00000362 a5a5a5a5 a5a5a5a5 a5a5a5a5
3fff7e70: 00000000 00000000 3fff8638 00000000
3fff7e80: 402206ee 00000362 3fff879b 00000001