espressif/ESP8266_RTOS_SDK #154: Not able to connect to the AP

Rajkumar181, 2018-03-13:
when i am dumping the code into esp8266 12e , is working fine. but the same code , dumping to the ESP8266 v1 it is not connecting to the AP.

compiling the code --
STEP 1: use boot_v1.2+ by default
boot mode: new

STEP 2: choose bin generate(0=eagle.flash.bin+eagle.irom0text.bin, 1=user1.bin, 2=user2.bin)
enter (0/1/2, default 0):

ignore boot
generate bin: eagle.flash.bin+eagle.irom0text.bin

STEP 3: choose spi speed(0=20MHz, 1=26.7MHz, 2=40MHz, 3=80MHz)
enter (0/1/2/3, default 2):

spi speed: 40 MHz

STEP 4: choose spi mode(0=QIO, 1=QOUT, 2=DIO, 3=DOUT)
enter (0/1/2/3, default 0):

spi mode: QIO

STEP 5: choose spi size and map
    0= 512KB( 256KB+ 256KB)
    2=1024KB( 512KB+ 512KB)
    3=2048KB( 512KB+ 512KB)
    4=4096KB( 512KB+ 512KB)
    5=2048KB(1024KB+1024KB)
    6=4096KB(1024KB+1024KB)
    7=4096KB(2048KB+2048KB) not support ,just for compatible with nodeMCU board
    8=8192KB(1024KB+1024KB)
    9=16384KB(1024KB+1024KB)
enter (0/2/3/4/5/6/7/8/9, default 0):
6
spi size: 4096KB
spi ota map:  1024KB + 1024KB

Flash command --- esptool.py --port /dev/ttyUSB0 write_flash 0x00000 eagle.flash.bin 0x20000 eagle.irom0text.bin 0x7c000 esp_init_data_default.bin 0x7e000 blank.bin

OUTPUT ---
�l��$`�rn����SDK version:2.0-dev(9ec59b5) 72272
Board Mac id: 6201940ed44b
mode : sta(60:01:94:0e:d4:4b)
add if0

27-301, 2018-03-15:
Some questions:
1. What is the flash size of your module?
2. Can you use `wifi_station_get_auto_connect` to check whether auto connect is enable?

Rajkumar181, 2018-03-15:
flash size is 1M.
And i will try this.

Rajkumar181, 2018-03-15:
i tried with this wifi_station_get_auto_connect  but still the same behaviour.

27-301, 2018-03-16:
For esptool, you don't set `--flash_size`, so it will detect the flash size auto. Since your flash size is 1MB, the parameter sector are default at 0xFD000~0xFF000, there are two methods to check, you can try one:
1. when program, add --flash_size=512KB for esptool, don't change the program address of each bin;
2. use default flash size, and program esp_init_data_default.bin to 0xFC000, and blank.bin to 0xFE000.

BTW, i let you try `wifi_station_get_auto_connect`, just want to check it's true or false, this cannot change anything.

Rajkumar181, 2018-03-16:
esptool.py --port /dev/ttyUSB0  --flash_size=512KB write_flash 0x00000 eagle.flash.bin 0x20000 eagle.irom0text.bin 0x7c000 esp_init_data_default.bin 0x7e000 blank.bin 

Like this i have to try.

27-301, 2018-03-27:
@Rajkumar181 Do you try? what's the result?

myco, 2018-04-16:
I have same issue with my wemos d1 mini. I Tried to write:
0x00000 eagle.flash.bin
0x10000 eagle.irom0text.bin
0x7c000 esp_init_data_default.bin
0x7e000 blank.bin

and also:
0x00000 eagle.flash.bin
0x10000 eagle.irom0text.bin
0xFC000 esp_init_data_default.bin 
0xFE000 blank.bin

Always get some mess on serial port.

FayeY, 2018-05-24:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

closed 2018-05-24 by FayeY

zdila, 2020-05-05:
I experienced simmilar problem in nonrtos-sdk on ESP8266 with DOUT flash. Setting 0xff000 to 0 and 0xff008 to 5 fixed the issue. Is data on that address documented somewhere?