espressif/ESP8266_RTOS_SDK #155: cannot run simple mqtt example after ESP8266_RTOS_SDK pre-release 2.0.0-rc1

MohammadAhmad1990, 2018-03-13:
SDK was working fine 2 days back but now after the new release I have compiled the code and start the serial monitor. And the controller seems to get stuck at MQTT connected. Kindly fix this ASAP.

`SDK version:2.0-dev(79c549f) 71664
mode : sta(2c:3a:e8:05:ec:59)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 9
pm open phy_2,type:2 0 0
cnt

connected with netis, channel 6
dhcp client start...
sta connected
ip:192.168.11.205,mask:255.255.255.0,gw:192.168.11.1
sta got ip ,create task and free heap size is 69280
mqtt client thread starts
Use MQTTStartTask
MQTT Connected`

llymaximus, 2018-03-14:
@MohammadAhmad1990 

Hi, I need some information here to debug this issue:  
1. did you just run the mqtt demo without any modifications (except the mqtt broker address & port)?
2. which mqtt broker did you connect with?

Thanks

llymaximus, 2018-03-14:
So which mqtt broker did you connect with？Does it work fine after restarting the demo？If the problem still exists, could you please share the generated binary files. Thanks

MohammadAhmad1990, 2018-03-14:
I have used Mosquitto broker. The problem still persists after the restart. The controller does not reset but it get stuck at MQTT connected.
Is there any other option of sharing the binaries other than uploading them on github ?

27-301, 2018-03-15:
Same issue when use `iot.eclipse.org` to do the test.

MohammadAhmad1990, 2018-03-15:
@27-301 I have not been able to publish messages successfully with the previous version of the SDK. If I modify the payload in the subscription callback function and publish it. The next time I receive message from the subscribed message is garbage. 
The messages from the subscribed topic are all ok until I dont publish anything. Can you help me in this matter

27-301, 2018-03-15:
Seems that this bug is introduced in commit: e8b824eb1c64aeba64b83b6dddfade89f7ec42de. Just revert api_msg.c back to old version, have a try
[liblwip.a.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/1815541/liblwip.a.zip)

MohammadAhmad1990, 2018-03-18:
Any update on this issue ?

27-301, 2018-03-27:
@MohammadAhmad1990 commit 408104dca7e27704bea18546d153a45dab7f9e76 is a fix for this, have a try.

MohammadAhmad1990, 2018-04-01:
@27-301 Sorry for the late reply. Ok sure I will and update

closed 2018-05-24 by FayeY