espressif/ESP8266_RTOS_SDK #156: UDP getting error or not able to send the packet

Rajkumar181, 2018-03-16:
I have enabled both Station mode and AP mode with UDP protocol, it was working fine i.e getting the packets on both (i.e. station mode as well as AP mode). If i disconnect my mobile from AP mode abruptly,  afterwards i am not getting data from both sides. UDP protocol API sento() is returning (-1).

27-301, 2018-03-16:
Can you provide demo code to reproduce this issue? also log. Otherwise, others need to write the code and try, this will cost time.

And if you disconnect your mobile normally? is it OK? How you disconnect your mobile from AP? power off the phone?

Rajkumar181, 2018-03-16:
yes i disconnected mobile normally , i just switch off my mobile wifi so it disconnected from ESP AP.

just now i found the reason  of this i trying to send the data to this IP 192.168.4.2.

Rajkumar181, 2018-03-16:
I written a sample code for testing purpose. 

This is console print --

connected with Manas, channel 11                                                
dhcp client start...                                                            
sta connected                                                                   
ip:192.168.2.31,mask:255.255.255.0,gw:192.168.2.1                               
sta got ip ,create task and free heap size is 48360                             
add 1                                                                           
aid 1                                                                           
station: c4:0b:cb:d6:09:55 join, AID = 1                                        
EVENT_SOFTAPMODE_STACONNECTED                                                   
station: c4:0b:cb:d6:09:55 leave, AID = 1                                       
rm 1                                                                            
EVENT_SOFTAPMODE_STADISCONNECTED                                                
Sending error on DHCP broadcast[1073705526]                                     
Sending error on DHCP broadcast[1073705526]                                     
Sending error on DHCP broadcast[1073705526]                                     
Sending error on DHCP broadcast[1073705526]                                     
Sending error on DHCP broadcast[1073705526]                                     
Sending error on DHCP broadcast[1073705526]                                     
Sending error on DHCP broadcast[1073705526]                                     
Sending error on DHCP broadcast[1073705526]

Testing code ----

/******************************************************************************
 * Copyright 2013-2014 Espressif Systems (Wuxi)
 *
 * FileName: user_main.c
 *
 * Description: entry file of user application
 *
 * Modification history:
 *     2015/7/3, v1.0 create this file.
*******************************************************************************/

#include "esp_common.h"
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "esp_wps.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"

#include "user_config.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"
#include "espressif/espconn.h"
#include "espressif/airkiss.h"

//#include "BoardConfig.h"
//#include "Relaycontrol.h"
#include "gpio.h"
//#include "eeprom.h"

#include "esp_wifi.h"
/*********** Global Declaration *********************/

bool wifiConnStatus = false;

bool  wiFiAPConnStatus = false;
void wifi_event_handler_cb(System_Event_t *event)
{
   
    if (event == NULL) {
[... 213 lines cut ...]
	uint8_t newmac[6];

	uart_div_modify(0, UART_CLK_FREQ / 115200);
	os_delay_us(1000);
	os_printf("SDK version:%s %d\n", system_get_sdk_version(), system_get_free_heap_size());
	wifi_station_config("Manas", "iramtech123");
	wifi_set_opmode(STATIONAP_MODE);

	//wifi_set_opmode(STATION_MODE);
	wifi_set_event_handler_cb(wifi_event_handler_cb); 
	wifi_station_connect();
	wifi_station_get_auto_connect();

	os_delay_us(1000);
	os_delay_us(1000);

	wifi_ap_config("ESP8266", "esp1234567");	

	
	xTaskCreate(UDP_Receive, "UDP_Receive", 2048, NULL, 2, NULL);
	xTaskCreate(UDP_Send, "UDP_Send", 2217,NULL,5,NULL );

	
    
}

Rajkumar181, 2018-03-16:
I written a sample code for testing purpose.

This is console print --

connected with Manas, channel 11
dhcp client start...
sta connected
ip:192.168.2.31,mask:255.255.255.0,gw:192.168.2.1
sta got ip ,create task and free heap size is 48360
add 1
aid 1
station: c4:0b:cb:d6:09:55 join, AID = 1
EVENT_SOFTAPMODE_STACONNECTED
station: c4:0b:cb:d6:09:55 leave, AID = 1
rm 1
EVENT_SOFTAPMODE_STADISCONNECTED
Sending error on DHCP broadcast[1073705526]
Sending error on DHCP broadcast[1073705526]
Sending error on DHCP broadcast[1073705526]
Sending error on DHCP broadcast[1073705526]
Sending error on DHCP broadcast[1073705526]
Sending error on DHCP broadcast[1073705526]
Sending error on DHCP broadcast[1073705526]
Sending error on DHCP broadcast[1073705526]

Rajkumar181, 2018-03-16:
Find the zip file.
[esp8266.tar.gz](https://github.com/espressif/ESP8266_RTOS_SDK/files/1818333/esp8266.tar.gz)

Rajkumar181, 2018-03-19:
Any update why it is happening , how we can fix the bug.

fafactx, 2018-06-05:
sta and ap mode is not full duplex，so you can use only one mode to test you demo at a time。

fafactx, 2018-06-05:
void user_init(void)
{
	uint8_t newmac[6];

	uart_div_modify(0, UART_CLK_FREQ / 115200);
	os_delay_us(1000);
	os_printf("SDK version:%s %d\n", system_get_sdk_version(), system_get_free_heap_size());
	wifi_station_config("Manas", "iramtech123");
	wifi_set_opmode(STATIONAP_MODE);

	//wifi_set_opmode(STATION_MODE);
	wifi_set_event_handler_cb(wifi_event_handler_cb); 
	wifi_station_connect();
	wifi_station_get_auto_connect();

	os_delay_us(1000);
	os_delay_us(1000);

	wifi_ap_config("ESP8266", "esp1234567");	

	
	xTaskCreate(UDP_Receive, "UDP_Receive", 2048, NULL, 2, NULL);
	xTaskCreate(UDP_Send, "UDP_Send", 2217,NULL,5,NULL );

	
    
}

in this pdf 20a-esp8266_rtos_sdk_programming_guide_en.pdf

tell us task stack should from 176 to 512 ，this not BYTE for a calc，reference freertos readme。
you can try this。

FayeY, 2018-12-20:
Sorry that we have refactored the SDK to be ESP-IDF style, would you like to have a try with our latest SDK? If the issue persists, please feel free to reopen or submit a new one. Thanks.

closed 2018-12-20 by FayeY