espressif/ESP8266_RTOS_SDK #158: SDK 2.0-dev crashes with simple program

THKDev, 2018-03-23:
Since [commit](https://github.com/espressif/ESP8266_RTOS_SDK/commit/eb91560c0c697381dbeb20d2fa8335e2952007a6) any program compiled with PlatformIO environment will crash. The program will run for 3 seconds, than hang and reboot. Before this commit same programs run fine without any problem.
What has changed with this commit?

Log from simple program which does nothing.

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30364, room 16
tail 12
chksum 0x21
ho 0 tail 12 room 4
load 0x3ffe8000, len 2008, room 12
tail 12
chksum 0x5e
ho 0 tail 12 room 4
load 0x3ffe87e0, len 568, room 12
tail 12
chksum 0x32
csum 0x32
mode: QI␏5
OS SDK ver: 2.0-dev(9ec59b5) compiled @ Feb 12 2018 11:02:32
phy ver: 1055_1, pp ver: 10.7

rf cal sector: 1019
tcpip_task_hdl : 40107a28, prio:10,stack:512
idle_task_hdl : 40107ad8,prio:0, stack:384
tim_task_hdl : 40107c20, prio:2,stack:512
entering user_init
leaving user_init
mode : null
..........␂@�Z-nP␙␕␓�M␌%��Z     %Y␎+Q
        �E␋XÈ␀��␂��I
␑\␓␑␑   )n@␑��B�[
␑␁!LO@␕)        K�T␑�␋�V␑1␎\␓��␏␑)�␑Z�
␑␑@�\␕␑ 5␏␓�␂�NRA␌␋�ؚ�P9J)�C␄1␌T␑��Y
␑␀�R␅␄R��␗�     )l�␕    5Y
␓�␂�NR�␕
ʳ␏␑␅    J�
␑Zq��Q
␁␌␎5
OS SDK ver: 2.0-dev(9ec59b5) compiled @ Feb 12 2018 11:02:32
phy ver: 1055_1, pp ver: 10.7
```
[sdktest.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/1843470/sdktest.zip)

Kind regards

27-301, 2018-03-29:
Can you try your code by using sdk directly? Not familiar with PlatformIO.

27-301, 2018-03-29:
@THKDev Can you show me how to use the newest SDK in PlatformIO? Is there some configurations in PlatformIO?

27-301, 2018-03-29:
Tried platformio with this demo, it halted then crash, very strange, by using SDK directly, it works well. From the commit message, it's `feat(lib): Use gcc to compile all internal libs`. It maybe relate with the Makefile used by platform, can you check with PlatfomIO guys?

FayeY, 2018-05-28:
Is there any update? Is this issue solved now?

THKDev, 2018-05-28:
I do NOT use Platform.IO anymore. In my option it is to buggy. Platform.IO [modifies](https://github.com/yanbe/framework-esp8266-rtos-sdk/blob/master/package.bat) some libraries from ESP8266_RTOS_SDK to get it working inside the Platform.IO environment.
I switch to the build process of ESP8266_RTOS_SDK and anything runs fine.

closed 2018-05-28 by FayeY