espressif/ESP8266_RTOS_SDK #159: 有人知道RTOS_SDK 的 boot 源代码开源了吗？想修改下默认波特率。

fafactx, 2018-03-27:
I'd like to revise the baud rate for the next boot. I don't know where the source code is？
Thanks for reply

27-301, 2018-03-28:
Not release now, which baud rate do you want to use?

dupeige, 2018-04-19:
yes,i gree

dupeige, 2018-04-19:
I want to use of arduino

dupeige, 2018-04-19:
我也想改变其比特率。想用它与arduino通信

Derekxiong, 2018-05-13:
I think the easy way is too change the baud rate in the arduino.

fafactx, 2018-05-18:
becasue of the ocs = 26M ,then defautl the boot v1.6 or bootv1.7 will run as baud rate = 74880，
My question is if the boot can open source for me to change the default baud rate.
板卡时钟是26M情况下，boot版本对应默认波特率是74880，这个是否可以开源，我需要一个更通用波特率。

FayeY, 2018-05-28:
Hi, so sorry that the boot is not open source, and the startup baud rate cannot be changed. You can only change the baudrate in user_init. Sorry for the inconvenience.

closed 2018-05-28 by FayeY