espressif/ESP8266_RTOS_SDK #160: RTOS libs 编译失败

jayyuz, 2018-03-28:
ETS_NMI_UNLOCK 
lib/libfreertos.a(port.o): In function `SoftIsrHdl':
(.text.SoftIsrHdl+0x47): undefined reference to `ETS_NMI_UNLOCK'
collect2: error: ld returned 1 exit status

27-301, 2018-03-29:
Do you follow this step? such as:
1. enter third_party, then run `./make_lib.sh freertos`
2. enter examples/openssl_demo, run `./gen_misc.sh`

Do you change some code or demo? If yes, can you show your change? or upload your code here. I can help to check.

ivanroberto, 2018-03-30:
Hi guys, 
Same here, I did 1 and 2 and still the same.
Using the mqtt_demo offered in example (without modifications)
Thanks

`xtensa-lx106-elf-gcc  -L/home/ivan/esp/esp8266/ESP8266_RTOS_SDK/lib -Wl,--gc-sections -nostdlib -T/home/ivan/esp/esp8266/ESP8266_RTOS_SDK/ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lcirom -lgcc -lhal -lcrypto -lfreertos -llwip -lmain -lnet80211 -lphy -lpp -lmbedtls -lopenssl -lmqtt -lwpa user/.output/eagle/debug/lib/libuser.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 
/home/ivan/esp/esp8266/ESP8266_RTOS_SDK/lib/libfreertos.a(port.o):(.text.SoftIsrHdl+0x8): undefined reference to `ETS_NMI_LOCK'
/home/ivan/esp/esp8266/ESP8266_RTOS_SDK/lib/libfreertos.a(port.o):(.text.SoftIsrHdl+0xc): undefined reference to `ETS_NMI_UNLOCK'
/home/ivan/esp/esp8266/ESP8266_RTOS_SDK/lib/libfreertos.a(port.o):(.text.SoftIsrHdl+0x17): undefined reference to `ETS_NMI_LOCK'
/home/ivan/esp/esp8266/ESP8266_RTOS_SDK/lib/libfreertos.a(port.o): In function `SoftIsrHdl':
(.text.SoftIsrHdl+0x47): undefined reference to `ETS_NMI_UNLOCK'
collect2: error: ld returned 1 exit status
/home/ivan/esp/esp8266/ESP8266_RTOS_SDK/Makefile:403: recipe for target '.output/eagle/debug/image/eagle.app.v6.out' failed
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
`

Governa, 2018-04-14:
Same problem here.
You may use this commit(in master branch) to solve it: e74ac4e6244f727497fe6a36a638b49981dc888b

FayeY, 2018-05-24:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-05-24 by FayeY