espressif/ESP8266_RTOS_SDK #161: How to use serial port interrupt in this SDK？

liefyuan, 2018-04-09:


liefyuan, 2018-04-09:
and when I 
void user_init(void)
{
    uart_init_new();
    if (xTaskCreate(main_process, "main_process", 1024, NULL, 5, NULL) != pdPASS) {
        print_error("main process");
    }
}
and 

LOCAL void
uart_intr_handler(void *para)
{
	// there is nothing
}

void
uart_init_new(void)
{
    UART_WaitTxFifoEmpty(UART0);
    UART_WaitTxFifoEmpty(UART1);

    UART_ConfigTypeDef uart_config;
    uart_config.baud_rate    = BIT_RATE_74880;
    uart_config.data_bits     = UART_WordLength_8b;
    uart_config.parity          = USART_Parity_None;
    uart_config.stop_bits     = USART_StopBits_1;
    uart_config.flow_ctrl      = USART_HardwareFlowControl_None;
    uart_config.UART_RxFlowThresh = 120;
    uart_config.UART_InverseMask = UART_None_Inverse;
    UART_ParamConfig(UART0, &uart_config);

    UART_IntrConfTypeDef uart_intr;
    uart_intr.UART_IntrEnMask = UART_RXFIFO_TOUT_INT_ENA | UART_FRM_ERR_INT_ENA | UART_RXFIFO_FULL_INT_ENA | UART_TXFIFO_EMPTY_INT_ENA;
    uart_intr.UART_RX_FifoFullIntrThresh = 60;
    uart_intr.UART_RX_TimeOutIntrThresh = 10;
    uart_intr.UART_TX_FifoEmptyIntrThresh = 20;
    UART_IntrConfig(UART0, &uart_intr);

    UART_SetPrintPort(UART0);
    UART_intr_handler_register(uart_intr_handler, NULL);  // handler is nothing
    ETS_UART_INTR_ENABLE();
}

in the serial assistant I got this 

>>tout
>>hello!!

Miracle !! the interrupt handler is nothing but the code can respond!! and very very horrible thing is a string "tout" ,My god ,that string where is come from????

xcguang, 2018-05-28:
Please check if your uart_intr_handler is called.

FayeY, 2018-06-06:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2018-06-06 by FayeY