espressif/ESP8266_RTOS_SDK #162: new SDK ShowCritical

leo-ivan, 2018-04-10:
hello everyone,
    I got system crashed when updating to the latest version(V2.0.0) which I used lwip to send/recv message. Do you know how this happen and can anyone explain the meaning of the following information for me?
ShowCritical:0
HdlMacSig:0
SWReq:0

27-301, 2018-04-13:
This means that software WDT is triggered. Can you share your test steps and your test code? and also more logs before this happened.

CaineLew, 2018-05-17:
I used recv message,and then call 'pwm_set_duty' and 'pwm_start',sometimes the system crashed.all the code run in task.

FayeY, 2018-05-24:
@CaineLew Please submit a new issue for your problem, and it will be very helpful if you could provide your test code so we can reproduce the problem and help debug it.

@leo-ivan , I'm going to close this issue due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

Thanks.

closed 2018-05-24 by FayeY