espressif/ESP8266_RTOS_SDK #164: Can't find gen_misc.sh

xiongyumail, 2018-04-23:
Where can I find the gen_misc.sh profile in examples? There are seems like some problems with this master branch.

27-301, 2018-04-24:
hi,
The documents should be updated.

From the current README:

Roadmap
ESP8266_RTOS_SDK's framework is quite outdated and different from the current esp-idf and we are planning to migrate ESP8266_RTOS_SDK to esp-idf eventually after v2.0.0.

However, we will firstly provide a new version of ESP8266 SDK (ESP8266_RTOS_SDK v3.0), which shares the same framework with esp-idf (esp-idf style), as a work-around, because the multi-CPU architecture is not supported by esp-idf for the time being.

27-301, 2018-04-24:
If you are familiar with esp-idf, then it will be simple.
Basic steps:
export IDF_PATH=~/esp/ESP8266_RTOS_SDK
make defconfig or make menuconfig, set you `Serial flasher config`
make
make flash
make monitor

27-301, 2018-05-18:
Update README in b32b40cf4424e146e50da9eb9a46e657bf37d50a

closed 2018-05-18 by 27-301

GaryHendrick, 2018-06-02:
The master branch still does not have this updated README within examples/get-started/project_template.  This causes confusion.