espressif/ESP8266_RTOS_SDK #165: ESP8266_RTOS_SDK-2.0.0 websocket_demo compil

rillhu, 2018-04-24:
Dear espressif,

When I compile the websocket_demo of RTOS_SDK V2.0, I meet many errors.

> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o):(.text.nopoll_base64_encode+0x0): undefined reference to `base64_encode'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o): In function `nopoll_base64_encode':
> (.text.nopoll_base64_encode+0x16): undefined reference to `base64_encode'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o):(.text.nopoll_cleanup_library+0x4): undefined reference to `EVP_cleanup'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o):(.text.nopoll_cleanup_library+0x8): undefined reference to `CRYPTO_cleanup_all_ex_data'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o):(.text.nopoll_cleanup_library+0xc): undefined reference to `ERR_free_strings'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o): In function `nopoll_cleanup_library':
> (.text.nopoll_cleanup_library+0x1e): undefined reference to `EVP_cleanup'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o): In function `nopoll_cleanup_library':
> (.text.nopoll_cleanup_library+0x24): undefined reference to `CRYPTO_cleanup_all_ex_data'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll.o): In function `nopoll_cleanup_library':
> (.text.nopoll_cleanup_library+0x2a): undefined reference to `ERR_free_strings'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll_conn.o):(.text.__nopoll_conn_ssl_verify_callback+0x0): undefined reference to `X509_STORE_CTX_get_current_cert'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll_conn.o):(.text.__nopoll_conn_ssl_verify_callback+0x4): undefined reference to `X509_get_issuer_name'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll_conn.o):(.text.__nopoll_conn_ssl_verify_callback+0x8): undefined reference to `X509_NAME_oneline'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll_conn.o):(.text.__nopoll_conn_ssl_verify_callback+0xc): undefined reference to `X509_get_subject_name'
> /mnt/c/Users/hongf/Work/04_wslShare/esp8266/ESP8266_RTOS_SDK-2.0.0/lib/libnopoll.a(nopoll_conn.o): In function `__nopoll_conn_ssl_verify_callback':
> (.text.__nopoll_conn_ssl_verify_callback+0x26): undefined reference to `X509_STORE_CTX_get_current_cert'

After I replace the libssl with the libmedtls and libopenssl in websocket_demo\Makefile
`old:    -lssl                   \`
```
-->
new:    -lmbedtls               \
new:    -lopenssl               \
```

And I also change the the include file in websocket.c as below:
`old:    #include "ssl_compat-1.0.h"`
-->
`new:    #include "openssl/ssl.h"`

Then I can compile the demo successfully. And it looks running OK.

So I think there are bug in this demo of RTOS SDK V 2.0.0, I hope Espressif can fix such error when release next version.
 Thanks!

27-301, 2018-04-24:
You can checkout branch `release/v2.x.x`, it has been fixed in commit cd1e3ffa306c771a218515f96931756f255b3fc8.

wujiangang, 2018-05-10:
Fixed in commit cd1e3ff

closed 2018-05-10 by wujiangang

sky9971, 2019-09-01:
Hi,
I'm using SDK version 3.2 can I also use this websocket demo example on the latest RTOS SDK for ESP8266